victronenergy/gui-v2 #564: /Settings/Gui/Brightness not created

kwindrem, 2023-11-23:
GUI v1 created com.victronenergy.settings /Settings/Gui/Brightness and set it's range.

It appears that nothing in the system creates this parameter if gui-v2 is running.

mpvader, 2023-11-24:
Hi @kwindrem , iirc its venus-platform that should do that now.

The new gui is no longer responsible for creating settings. Also any commands that require direct access to the OS, like reboot, or firmware updates, are handled via venus-platform rather than direct between UI and other scripts and services.

did you test this? Ie are you sure that setting isn’t created after disabling the old gui, restting to factory defaults and then rebooting?

kwindrem, 2023-11-24:
I didn't do a complete reset to factory default settings but I switched to the beta GUI, stopped local settings and edited settings.xml to remove /Gui/Brightness then rebooted. /Gui/Brightness did not return.

Might be other /Gui/... parameters also.

mpvader, 2023-11-25:
Hey Kevin, looks like you are right. gui-v1 still creates that setting as well as many more:

https://github.com/victronenergy/gui/blob/master/src/ve_qitems_settings.cpp

Note that (most) that most likely shouldn't be moved to gui-v2, but rather moved elsewhere. @jhofstee might already have a plan for this (?).

mpvader, 2024-01-10:
@ReinvdZee , I spoke to Jeroen about this, and the command to create that setting (as well as others) will need to moved to venus-platform, together with the related functionality discussed in #375

mpvader, 2024-01-10:
I've made a new issue to handle all required moving of settings, which is here: https://github.com/victronenergy/venus/issues/1197

So this one can be closed.

closed 2024-01-10 by mpvader