victronenergy/gui-v2 #567: Overview Page: Battery connection line is not animated

mr-manuel, 2023-11-24:
The battery connection line on the overview page is not animated. I tried with v0.1.20 and the latest main build.

[image]

chriadam, 2023-11-28:
@blammit if I remember correctly you spoke to jperez about this / the flow state is defined by some systemcalc value?

mr-manuel, 2023-11-28:
There is a threshold that has to be reached, that the flow is shown. I think it was 30 W.

blammit, 2023-11-28:
@mr-manuel  Is this regarding the flow between the Inverter/Charger and the Battery? I just checked and the animation seems to be working, so maybe something else is going wrong here. That is defined by whether the /Dc/0/Power of the main VE.Bus service (as defined by `com.victronenergy.system/VebusService`) is over 30 (or under 30). Could you check that value?

Also if this is being tested via wasm, there are some fixes currently underway involving the MQTT connection and data fetching, so that could also be a factor.

mr-manuel, 2023-11-29:
@blammit yes amd yes, it's the wasm build. I will try the latest main build or the branch, if I find one.

chriadam, 2023-11-29:
No MQTT fixes are available yet, I was initially hoping to have a PR up tonight but it looks like it won't be ready until tomorrow night.

mr-manuel, 2023-11-29:
No problem

chriadam, 2024-01-31:
This is still problematic and needs to be fixed.  Marking as medium_prio and giving 1.0.0 tag.

DanielMcInnes, 2024-01-31:
still an issue
[image]

PR #783 (2024-02-21): Re-enable animation between battery and inverter/charger
Fixes #567.

Not sure if this is a correct fix, but after the change System.qml and SystemImpl.qml uid paths start to match and the animation starts flowing on both the Ekrano (demo mode) and mocked desktop,
commit c675e661 (2024-02-21): Re-enable animation between battery and inverter/charger

Fixes #567.
closed 2024-02-21 by jpetrell