victronenergy/gui-v2 #573: Language does not reflect the language setting value / check language list is up-to-date

blammit, 2023-11-27:
Reported issue is that if you set a non-English language in gui-v1, the selection is not reflected in gui-v2, and gui-v2 remains in English.

Also check that the language list is up-to-date, and has all the languages supported by gui-v1.

DanielMcInnes, 2023-11-27:
No translations have been done for gui-v2, so it is expected that English is displayed, no matter what language is selected.

kwindrem, 2023-11-27:
Would language be one of those things that is specific to one display? That is, one remote GUI could display in French and another German?

mpvader, 2023-11-27:
@kwindrem no.

blammit, 2023-11-28:
@DanielMcInnes the UI should still show the selected language, even if translations are not yet available.

DanielMcInnes, 2023-11-28:
I just tested this. I ran gui-v2 on my desktop, and connected via mqtt to gui-v1 running on my cerbo. If I changed the language to 'French' via gui-v2, I see the language change to 'French' on gui-v1 on the cerbo. I then changed the language from 'French' to 'English' on gui-v1 on the cerbo, and I saw the selected language change on gui-v2 on my desktop.

chriadam, 2023-11-28:
(Note that the only way you will be able to tell if the language in gui-v2 is actually changed, is if you change to French and then check the text of the Levels button in the navigation bar - as it is the only string we have translated - to "Niveaux".)

Thanks for checking that, Daniel.

One other point: on Slack someone asked if we could ensure that the latest languages supported in gui-v1 are also available in gui-v2 (e.g. Ukrainian was recently added to gui-v1, but we probably haven't added that one to the Languages list in gui-v2 yet).

chriadam, 2024-01-31:
Open PR for this one: https://github.com/victronenergy/gui-v2/pull/673

closed 2024-02-08 by blammit