victronenergy/gui-v2 #574: Controls Card view enter transition animation should be "from top + fade"

chriadam, 2023-11-28:
Currently, the control cards view animates in from the left rather than the top.
It should slide down from the top (and fade in as it animates).
The view "below" it should NOT move, but just be faded out "below" it.
That is, the controls card view is kind of an "overlay" view, rather than a true "navigation page".
See discussion on Slack with Serj for the Figma prototype.

blammit, 2023-12-08:
Could also update the control cards open/close icon as part of this task. When it is "open", the icon should be highlighted. See current Figma design.

chriadam, 2024-01-31:
Simple task, needs to be done for 1.0.0.  Setting as medium_prio.

chriadam, 2024-01-31:
The icon task Bea mentioned is tracked separately as #584

jpetrell, 2024-02-12:
Draft PR https://github.com/victronenergy/gui-v2/pull/744

[access-controls-from-top.webm](https://github.com/victronenergy/gui-v2/assets/2203667/0ebb81d1-d9eb-4a6a-9dda-cbb78a65c03e)

Didn't update any icons yet. I will ask Serj for feedback.

commit 449a7da8 (2024-02-14): Access controls from top

Fixes #574.
closed 2024-02-14 by jpetrell