victronenergy/gui-v2 #575: Battery "no further discharge allowed" state needs to be visualised in Brief+Overview pages

chriadam, 2023-11-28:
New designs / new feature requirement.

In some setups, user can set a setting which specifies that when the battery reaches a certain discharge level, no further discharge from the battery is allowed (and instead, all current drawn must come from the grid - presumably to protect service life of battery / prevent full discharge).

Figma has new information about how this state should be visualized on the Brief and Overview page.

chriadam, 2024-01-31:
Discussed that this might not be needed for 1.0.0 release, but should be considered "nice to have for 1.0.0".
Removing 1.0.0 flag and setting to medium_prio.

jpetrell, 2024-04-24:
From Serj understood that there were some additional design requests, and that it is probably better to handle/improve more states at once in addition to the battery "no discharge allowed" state. Maybe this should be set to a blocked status. Anyway, check with Serj before implementing.