victronenergy/gui-v2 #576: UI horizontal dimensions wrong on 7inch device

chriadam, 2023-11-28:
Round UI elements (icons, circular gauge) are stretched horizontally on 7inch device, according to Serj's testing.
Probably a wrong constant in the .json.
Might also be worth double checking that the scaling I added for WebAssembly is definitely disabled for CerboGX etc?

chriadam, 2024-01-31:
High prio.

chriadam, 2024-02-02:
This is a curious one.
When run on desktop, in seven-inch mode, gauges are perfectly round.
When run on CerboGX with 7-inch screen, gauges are 5mm wider than they are high (60 mm wide, 55 mm high).

But in both cases, Qt/QML "thinks" that they are round.  That is, I placed a round circle in the background of the gauges which is red, its width is equal to its height (in QML) and its radius is half the width.

My first suspicion is that the actual screen's pixels aren't square?
But I haven't yet totally ruled out other possibilities.

[image]
[image]

See the attached images.

jpetrell, 2024-04-30:
Also reproduces on Ekrano GX's 7" screen (circle is 60 mm wide, 55 mm high). We can apply a correcting horizontal scaling to the whole GUI v2, but then would need to decide what to do with the extra 8% or 85 pixels of horizontal space on the screen (see thin red borders on the picture below). Is the skewed dimensions seen an issue on all the views (e.g. text appearing wider than it should be)? If not, could also choose to apply the correction only to the circle. Personally, I am leaning towards closing as won't fix to avoid introducing an extra UI variant to test and maintain.

```
transform: Scale { origin.x: parent.width/2; origin.y: parent.height/2; xScale: 55.0/60.0 }
```

[image]

jpetrell, 2024-06-06:
> Are the skewed dimensions seen as a serious issue for all the views (e.g. text appearing wider than it should be)? If not, could also choose to apply the correction only to the circle. Though personally I am leaning towards closing as won't fix to avoid introducing an extra UI variant to test and maintain.

No reply. Clearly not for 1.0.0, dropping the milestone target.