victronenergy/gui-v2 #583: Screen Sleep function / feature

chriadam, 2023-11-30:
See https://victrondevelopment.slack.com/archives/C020QG3JVEW/p1701166638589059

Add a new button to the top-right of the brief page (actually the status bar) which puts the screen to sleep (not just black, but turn off backlight).  This allows people to go to sleep if their unit is in a caravan etc.

Needs support from venus-platform for hardware hooks (display on/off) if that doesn't exist already.

Once a screen is asleep, any touch input on the screen should wake it up again.

blammit, 2023-12-07:
> Needs support from venus-platform for hardware hooks (display on/off) if that doesn't exist already.

Doesn't exist yet; marking this task as blocked. Will need something like the screen-blanking and wake-up features in gui-v1 (implemented by VePlatform/VePlatformVenus).

blammit, 2023-12-11:
Discussed with Mathijs, this feature shoudn't need to be in venus-platform. We can implement this according to whatever gui-v1 does, and put the implementation in gui-v2, or implement in venus-platform if needed.

Note: the sleep button shouldn't appear in wasm builds as we cannot turn the display/backlight off there. It should only appear in device builds.

chriadam, 2024-01-31:
Marking as high priority.

mpvader, 2024-02-05:
Lets move this one out of v1.0.0. Its oone of the urgent things after first release, but should definitively not block it.

commit ce750a99 (2024-03-19): Implement display off button and timer

Fixes #859. Fixes #583.
commit 7874b382 (2024-03-19): Implement display off button and timer

Fixes #859. Fixes #583.
commit cd0bb12b (2024-03-20): Implement display off button and timer

Fixes #859. Fixes #583.
commit 43098d6d (2024-03-20): Implement display off button and timer

Fixes #859. Fixes #583.
closed 2024-03-22 by jpetrell