victronenergy/gui-v2 #588: Used fixed width containers for QuantityLabel

blammit, 2023-12-04:
This avoids issues where adjacent components constantly shift in alignment when the numbers change in the QuantityLabel.

See gui-v1 and what it uses for fixed widths.

blammit, 2023-12-05:
This will probably need to be customized depending on the font size and the unit shown in the QuantityLabel.

See related issue: https://github.com/victronenergy/gui-v2/issues/479

chriadam, 2024-01-08:
That PR 593 was merged, however after some testing, Serj raised some concerns with it, which we need to address.

Currently, the method we use here is to:
a) calculate the number of digits in the number (e.g. 1000 kW has 4 numeric digits)
b) allocate (digits * widthOfWidestGlyph) pixels for the label (i.e. fixed width)
c) render the text within that label as per normal

This solves the basic problem (of the label width changing every time the value changes) but it introduces new problems: since the widest glyph is 4, a number like 1111 will be much slimmer than 4444.  Thus, the label will be vastly "oversized" (using normal text rendering), and so if the text is Left Aligned or Center Aligned then there will be a large gap between the number and the units label (kW), and the gap will change depending on the specific value.  If the label is Right Aligned then in various parts of the UI the labels won't be properly aligned to each other (e.g. the left edge of 4444 won't align with the left edge of 1111).

Serj has a potential solution: render the numbers as monospaced glyphs into images, which can be cached and displayed.

This requires the following steps:

1) Create a QQuickImageProvider registered for "digits://" scheme
2) the path it supports will be the digits [0..9]
3) it will support query parameters for specifying the font pixel size, boldness, and color values
4) it will use QPainter to draw the glyph into a texture whose size is the size of the largest glyph (i.e. 4), and it should center the glyph within that texture (so 1 will be centered within the box)
5) then, QuantityLabel effectively becomes:

```
Row {
    Repeater {
        model: numberOfDigits
        delegate: Image {
            required property int index
            property int which: text.length - index - 1
            source: "digits://" + value.toString().at(which) + "?fontSize=" + font.pixelSize + "&bold=" + font.bold + "&color=" + textColor.toString()
        }
    }
}
```

This will mostly solve the problem, and should be reasonably fast (since the glyphs will be shared).

A few points of note:

1) we might need a special case for the period/dot glyph, otherwise the space around it will look too big (so, don't render it into a texture whose size matches the size needed to render '4' but instead just a small texture).
2) we need to be careful not to change "numberOfDigits" often, because otherwise the delegates will be recreated each time, which will be very slow.  The current code should be fine in this regard (i.e. numberOfDigits is an intermediate property bound to `text.length` which changes infrequently).  There is still a "bad case" where it flips between e.g. 999 and 1000, but maybe we can solve this with something like:

```
// latch the largest number of digits, so that we don't recreate delegates constantly
// if the value flips between 99 and 100 a lot, for example.
property int numberOfDigits: Math.max(numberOfDigits, text.length)

...

// then within the Row's Repeater, if we have allocated more images than we have
// digits to display, ensure that the unused images don't display any digit.
delegate: Image {
    source: {
        required property int index
        property int which: text.length - index - 1
        source: which < 0 ? "" : "digits://" + value.toString().at(which) + "?fontSize=" + font.pixelSize + "&bold=" + font.bold + "&color=" + textColor.toString()
    }
}
```

Finally, note that there still will be some "difference" in RHS padding if the last digit is 1 instead of 4 (because 1 will be centered), but much less than currently.

chriadam, 2024-01-12:
Note: as mentioned by Matthijs, please double check whether we can't just edit the font directly to monospace all of the numbers/digits.  Might be simpler.  But check that we have the license/rights to do that, etc.

closed 2024-01-25 by blammit