victronenergy/gui-v2 #606: Device List: com.victronenergy.genset.xxx state issue

philipptrenz, 2023-12-13:
On the `com.victronenergy.genset.xxx` device list item, the `Status` value is translated/mapped incorrectly.

[image]
[image]

philipptrenz, 2023-12-13:
Correct mapping is documented at https://github.com/victronenergy/venus/wiki/dbus#generator-data

```
/StatusCode                 <- 0=Standby; 1..7=Startup; 8=Running; 9=Stopping; 10=Error
```

blammit, 2023-12-21:
Thanks for reporting, will be fixed by https://github.com/victronenergy/gui-v2/pull/626

closed 2023-12-21 by blammit