victronenergy/gui-v2 #610: Rewrite Theme to avoid run-time json loading

chriadam, 2023-12-15:
Currently, the Theme values are loaded at run-time, and exposed in nested QQmlPropertyMaps.
This means that the compiler cannot do any optimisations, as the "available properties" aren't known at compile time, nor the property type, etc.

Instead, we should write a small tool which generates C++ code at compile time from the .json, and produces as output C++ classes with specific, well-typed properties.

e.g.:

```
class ThemeClass : public QObject
{
    Q_OBJECT
public Q_SIGNALS:
    void propertiesChanged(); // QML change signal.
    void formFactorChanged(int formFactor);
public Q_SLOTS:
    void handleFormFactorChanged(int formFactor) {
        if (m_formFactor != formFactor) {
            m_formFactor = formFactor;
            formFactorChanged(formFactor);
        }
    }
private:
    int m_formFactor = 7;
};

class ThemeGeometryScreen : public ThemeClass
{
    Q_OBJECT
    QML_ELEMENT
    Q_PROPERTY(qreal width READ width NOTIFY propertiesChanged FINAL)
    Q_PROPERTY(qreal height READ height NOTIFY propertiesChanged FINAL)
public:
    qreal width() const { return m_formFactor == 7 ? 1024 : 800; }
    qreal height() const { return m_formFactor == 7 ? 600 : 480; }
};

class ThemeGeometry : public ThemeClass
{
    Q_OBJECT
    QML_ELEMENT
    Q_PROPERTY(ThemeGeometryScreen *screen READ screen CONSTANT FINAL)
    Q_PROPERTY(ThemeGeometryPage *page READ page CONSTANT FINAL)
    ...
public:
    ThemeGeometry() {
        connect(this, &ThemeGeometry::formFactorChanged,
                     &m_screen, &ThemeGeometryScreen::handleFormFactorChanged);
        connect(this, &ThemeGeometry::formFactorChanged,
                     &m_page, &ThemeGeometryPage::handleFormFactorChanged);
        ...
    }
    ThemeGeometryScreen *screen() const { return &m_screen; }
    ThemeGeometryPage *page() const { return &m_page; }
    ....
private:
    ThemeGeometryScreen m_screen;
    ThemeGeometryPage m_page;
    ...
};

class Theme : public QObject
{
    Q_OBJECT
    QML_ELEMENT
    QML_SINGLETON
    Q_PROPERTY(ThemeGeometry *geometry READ geometry CONSTANT FINAL)
    Q_PROPERTY(ThemeTypography ...)
    ...
};
```

chriadam, 2023-12-18:
https://github.com/victronenergy/gui-v2/pull/613

chriadam, 2023-12-18:
Ideally it would be a python script which would run automatically as part of the build process, but this is fine for now.

chriadam, 2023-12-21:
See https://github.com/victronenergy/gui-v2/issues/625 for the task tracking rewriting themeparser as a python script

chriadam, 2023-12-21:
In future, we should change the theme .json files so that instead of . separators, they use _ separators, that way we can search the source for the exact properties to check whether they are used in the codebase or not.  (This will mean that we will ONLY support flat-mode theme object generation, not nested; but for performance flat-mode is far better, anyway.)

blammit, 2024-01-16:
Done now, closing.

closed 2024-01-16 by blammit