victronenergy/gui-v2 #611: Usage from grid off by factor 10?

mpvader, 2023-12-17:
https://community.victronenergy.com/questions/249509/venus-os-v32034-available-for-testing.html

[image]

DanielMcInnes, 2023-12-18:
I can't easily reproduce this, I see the following:
[image]
[image]

DanielMcInnes, 2023-12-18:
I can reproduce this now, it happens when we go over 9999W

chriadam, 2023-12-18:
Possibly a bug in my recent Units.js to .cpp conversion?

closed 2023-12-18 by DanielMcInnes