victronenergy/gui-v2 #614: Device List: Battery page issues

mr-manuel, 2023-12-18:
## Details page

[image]

## IO page

System Switch value does not exist in DBUS and should not be visible therefore.

[image]

## Device page

[image]

I will fix the errors by opening a PR.

mr-manuel, 2023-12-18:
@mpvader has the firmware always to be a hex number? If yes, then I have to change the versioning format of all my add-ons :-)

https://github.com/victronenergy/gui-v2/blob/8e57400d3fc631aa55f794cbd0cd62ae1ab29a1d/components/settings/ListFirmwareVersionItem.qml#L28-L40

I searched FirmwareVersion on https://github.com/victronenergy/venus/wiki/dbus-api and it says that it can be a string or did I missunderstood that?

> /FirmwareVersion
> GetText: A string, containing the firmware version which is formatted equal as used in change logs of the product, and/or on labels etcetera. This string is also what the Venus GUI and VRM portal will show to users.
> 
> GetValue: Numeric representation of the firmware version. To be prepared for automatic firmware updates, make sure that the numeric value is such that a newer version leads to a higher number. And also that the number is in the same format in the firmware update files (if available for this product); so that when indexed in a library, the number can be used to check up-to-dateness of the product. All downstream software will not, and may not, format this number itself for showing the user. Instead it must use the string as returned by GetText.

mr-manuel, 2023-12-18:
It it's ok, I would change

https://github.com/victronenergy/gui-v2/blob/8e57400d3fc631aa55f794cbd0cd62ae1ab29a1d/components/settings/ListFirmwareVersionItem.qml#L30-L31

to

```qml
		// Return the version as it is, if format is "vebus" or if the version is not a hex number
		if (format === "vebus" || !version.match(/^[0-9A-F]+$/i))
			return "v" + hexString;
```

mr-manuel, 2024-01-10:
With #616 I nearly resolved all problems. There are two problems left, that I was not able to solve.

1. Cell voltage shows only one decimal instead of three
2. The cell ID's for `minVoltageCellId`, `maxVoltageCellId`, `minTemperatureCellId` and `maxTemperatureCellId` are not shown, even if they are present on the debug page.
    [image]

@blammit could you check that?

blammit, 2024-01-11:
Sure, will look into the cell IDs issue next week. I also have an upcoming PR that will address the decimal issue.

mr-manuel, 2024-01-24:
> Sure, will look into the cell IDs issue next week. I also have an upcoming PR that will address the decimal issue.

Any news for this?

blammit, 2024-01-29:
Sorry, I forgot about this last week. Will definitely take a look this week and let you know.

blammit, 2024-01-31:
@mr-manuel can you see whether https://github.com/victronenergy/gui-v2/pull/698 fixes the issue?

mr-manuel, 2024-01-31:
Looks good now!

[image]

blammit, 2024-01-31:
Great, thank you -- will close as fixed.

closed 2024-01-31 by blammit