victronenergy/gui-v2 #618: Ethernet status shows "unplugged" when ethernet connection is manually configured

blammit, 2023-12-19:
Reported issue: in ethernet settings, ethernet status shows "unplugged" when ethernet connection is manually configured. 

Status displays correctly in gui-v1.

mr-manuel, 2023-12-19:
Also the WiFi status is wrong. It tells me that there is `No WiFi adapter connected`, even if I'm connected via WiFi.

chriadam, 2024-01-31:
Are these settings exposed by venus-platform?  i.e. are there appropriate MQTT paths for this too?

gui-v2 should not be using connman directly to interact with these settings.
Please double check this.

Marking as high_prio.

ReinvdZee, 2024-02-07:
See victronenergy/venus#1215. Settings will be exposed to dbus by venus-platform

chriadam, 2024-02-26:
Still waiting on platform enablers.  Adding "blocked" label.

ReinvdZee, 2024-02-28:
@blammit, @chriadam, @DanielMcInnes, @jpetrell, @jhofstee

These are the proposed changes to venus-platform in order to expose network settings to dbus/mqtt:

Venus-platform will relay connman's services to dbus/mqtt for the gui-v2 to work with. Venus-platform will not store or maintain settings.

Paths will be under `com.victronenergy.platform`

- 1 dbus path holds a json object with all available services (ethernet and WiFi  networks)
- 1 dbus path is used to connect to send network control commands to venus-platform (which parses and forwards to connman)
  - e.g: `/Network/SetValue` = `"{"network":<ssid>, "wpa":<wpa>, "action":"connect"}"`
- A few dbus persistent dbus paths hold the status of the connected network.
  - `/Network/Ethernet/State' = `connected`
  - `/Network/WiFi/State` = `connected`
  - `/Network/WiFi/SignalStrength` = `<signal strength>`
  - etc. 

Tasks for the gui:
- Parse the json object containing all ethernet and WiFi services. This can be done periodically when a page interested in these values is opened in the gui. 
- Use `Network/SetValue` to control the desired network. Details on what field are supported will follow. 
- Use `/Network/WiFi/SignalStrength` to display the correct WiFi connection symbol without having to reparse the json object repeatedly. 
- Use other fixed paths (like `/Network/WiFi/State`) to act on. Open for discussion. Parameters/Settings requiring continuous updates probably deserve a fixed path to prevent continuously reparsing the JSON object. 

I will work on the implementation and give a heads-up when ready

jpetrell, 2024-04-11:
While rewriting the network settings implemented few UI improvements #934 agreed with Serj, also fixed a bug with the Wi-Fi access point switch #935. Wi-Fi network settings are still quite limited, filed internal tasks about supporting hidden network, WPA2-Enterprise and captive portal logins.

jpetrell, 2024-04-15:
Setting the blocked label. Pending the integration of venus-platform changes before can be merged.

commit ceb8d09a (2024-04-09): Migrate network settings to venus-platform

Fixes #618.
commit 38e04c7a (2024-04-09): Migrate network settings to venus-platform

Fixes #618.
commit 5d184677 (2024-04-09): Migrate network settings to venus-platform

Fixes #618.
commit 2d09af97 (2024-04-09): Migrate network settings to venus-platform

Fixes #618.