victronenergy/gui-v2 #620: Scale kWh to Megawatt and Gigawatt if needed

blammit, 2023-12-19:
E.g. when displaying yield for a solar charger or PV inverter, using kWh may result in a very big number, so it would be nice to scale it further.

kwindrem, 2023-12-19:
Could use mega-watt-hours (MWh) 1/1,000,000. But that's a big jump. I don't see anything in-between 1/1,000 and MW.

Note mWh is the abbreviation for milii-watt-hours (1,000/1). Lower case for factors less than unity, capital letters for factors above unity.

So this leads to an inconsistency with kWh. It used to be that KW was the abbreviation for kilo-watt, but I now see kW or even kw is acceptable.

mr-manuel, 2023-12-19:
KW is wrong. If you follow the normed units then kW is correct. See https://www.bipm.org/en/measurement-units/si-prefixes. This is since 1960.

warwickchapman, 2023-12-19:
Largest Victron System currently supported is 15 x 12 = 180kVA ~ 144kW.

The HS19 15kVA will be followed soon by a 30kVA (with larger units planned) and will support many units in parallel, so lets assume we get above 1000kW in a single system (45 units of 30kVA > 1000kW).

1000kW is 1MW.

I would prefer to read 1089kW than 1.1MW.

[image]

When it comes to PV production, I would prefer to read 1089kW than 1089195W or 1.1MW.

kwindrem, 2023-12-19:
1.1 MW is soooooo much more readable than 1089kW.

I think the original issue was that on a large PV array, the production could exceed 10,000kWh or even 100,000kW. Showing separators helps but I'd still prefer shifting to 10 MWh and 100 MWh.

And once again, a space between the number and the unit really helps readability. This is especially true with a capital letter for the unit: 100W vs 100 W 1.1MWh vs 1.1 MWh.

mr-manuel, 2023-12-19:
If you wanna follow the "The International System of Units (SI)", then you have to place a space between the number and the unit. I would definitely recommend that.

The mega and giga factors are not for currect power production but for total power production. If you produce 100 kWh a day, then in a few years the number will be huge and much better readable by scaling it.

If I check the GUI then there is not much space to display four digits before the decimals except you make the column larger:

[image]

Maybe you create a new option to let the user decide to scale when reaching four (`>=1.000`) or five (`>= 10.000`) digits.

> When it comes to PV production, I would prefer to read 1089kW than 1089195W or 1.1MW.

@warwickchapman you will read 1,09 MWh, since there are two digits for decimals.

warwickchapman, 2023-12-19:
Ok of course. I had not thought of GWh.

Agree that 1.09MW works better than 1089kW.

kwindrem, 2023-12-19:
When I implemented kilo-units in GuiMods, I some people complained that they wanted power in units regardless of the size. Also some wanted the threshold to be adjustable. I would argue against this BUT provide some hysteresis in switching from MW to kW to W to avoid the unit changing often if the parameter is right at the threshold.

I would argue in favor of a "sane" number of significant digits. Showing 4 significant digits is probably way more than the precision of the measurement so showing 4.052 kW makes no sense. 3 significant digits is probably overkill based on measurement precision and user's need to differentiate values, but makes for a reasonable jump from unit so kilo-units, etc: 999 W to 1.01 kW

chriadam, 2024-01-31:
Wanted for 1.0.0, marking as medium_prio.

jpetrell, 2024-02-13:
> Also some wanted the threshold to be adjustable. I would argue against this

Yeah me too, adding configurability increases the overall maintenance burden, would only add such threshold settings later if enough customers raise it as an issue.

> provide some hysteresis in switching from MW to kW to W

Hysteresis adds a bit of complexity as then need to keep track of the previous values, wonder how big issue the oscillation between the formatting thresholds is in practice? AFAIK the scaling was mostly needed for formatting cumulative figures, which will only pass the thresholds once.

jpetrell, 2024-02-13:
PR https://github.com/victronenergy/gui-v2/pull/745

kwindrem, 2024-02-13:
Hysteresis is important. Any value close to the threshold would rescale constantly.

Keeping track of previous values is probably not necessary. Just keep track of the time the current value is less than the threshold and rescale once the time has elapsed. Switching to a larger unit (e.g., units to kilo-units) should happen immediately or the displayed value could exceed available space.

kwindrem, 2024-02-13:
Waiting until 10,000 to switch to kilo is hard to read without separators. (9999.9 W vs 9.99 kW). If you keep the precision, I think you really need separators: 9,000.0 (US) 9.000,0 in other areas of the world.

jpetrell, 2024-02-14:
> Waiting until 10,000 to switch to kilo is hard to read without separators.

Do you have a better threshold to suggest? AFAICS 10k threshold has simply been chosen to not go over the available number of digits.

If we start adding separators those start taking space and affecting the limits. There is currently no localization done on the value formatting. I am not against the separators, just the scope is growing a bit.

> Keeping track of previous values is probably not necessary. Just keep track of the time the current value is less than the threshold and rescale once the time has elapsed.

Well the main concern for me is having to carry state with each formatting call, whether that is related to value history or the time doesn't really matter. But since it is seen as crucial feature I can add it no problem. If somebody has good prior art implementation to share that would be appreciated so I don't invent the wheel . E.g. do you have any idea what would be good time filter value? I don't know how often GX hardware usually receives sensor updates. What if the value drops considerably? In that sense having two thresholds might be easier to create a middle region where the scales are not updated, e.g. switch to a larger scale when going over 10000, but don't switch back before going under 9000.

One thing that needs thinking is the precision. Currently cubic meters are displayed with 3 digit precision, kWh uses 2 digit precision, and volts, volt-amperes, amperes, hertz, ampere-hours use 1 digit precision, and litres, gallons, watts, W/m2 uses use 0 precision. These precisions I guess have been chosen because e.g. cubic meter is a large unit and hertz and litres are small units. But if the value ends up being large and needs unit scaling to kilos or megas I guess the existing precisions don't work that well any more.

mpvader, 2024-02-14:
Precision also depends in what it is.

for dc voltage, we don’t want three decimals precision because that implies super accurate (and unnecessary) voltage measurements. Which they are not.

for amps its ever more the case, only for very few cases / victron products there is more precision than one decimal for amps measurements.

I think Jesus has good experience in how to handle this switching between kW and W and MW etc. also in other gui’s; I’ve asked him.

mr-manuel, 2024-02-14:
> for dc voltage, we don’t want three decimals precision because that implies super accurate (and unnecessary) voltage measurements. Which they are not.

For the cell voltages we need three decimals, but they won't be scaled. Only in case this is relevant.

kwindrem, 2024-02-14:
I would have expected the threshold to kW to be at 1000 or somewhere close to that. But as soon as you display 1000 you might as well display 9999.

In GuiMods, the default threshold for kW is 1,000. Shortly after implementing that, I was asked to implement an adjustable threshold. It seems many people wanted to see 100000 watts rather than 100 kW. I am NOT suggesting an adjustable threshold or the ability to disable scaling. 

I think it's essential to keep the number of digits to something readable. Separators help but I understand why you'd not want to add that complexity since it's based on localization, or take the space needed. 3 significant digits is ideal for readability. 4 is probably the maximum.

Scaling should attempt to maintain the number of signifiant digits: 1.00, 10,0, 100, 1.0 kW ... Limiting power for example to 1 W seems excessive. But I can also see where accuracy may not support the precision.

cubic-meters is a odd one for capacities typically found in this kind of system. I thought about scaling down below unity but it seems the next step is to cubic-centimeters (a factor of 1,000,000) so that might not help. There is a cubic-decimeter  which is a factor of 1000 from cubic-meters but would anyone know what dm3 is ????

What is the real threshold for battery cell imbalance? Is 4 significant digits really necessary or just there because that's what people read on there volt meters?

From discussing this with users/system integrators I got the impression that 12345 provided more precision than 12.3 kW. While that's true, as mvader says, accuracy of the measurement makes anything past 2 or 3 digits meaningless.

I do not see a way to prevent scaling chatter without keeping some prior value or state. Separate scale up and scale down thresholds should prevent most scaling chatter nicely, but you would still need to keep the last scale factor around for the next measurement. I'm less concerned about large value changes than I am about sitting right at the threshold and having the display switching between W and kW every second.

jpetrell, 2024-02-20:
- Introduced new QML element `QuantityInfo { value: myValue; unitType: VenusOS.Units_Energy_KiloWattHour }` that implements hysteresis by remembering the existing formatting scale
- Larger scale is used whenever 10000 is passed, lower scale limit is 10% smaller at 9000 to avoid random cases where the value happens to oscillate close to the 10k threshold
- Currently the hysteresis is enabled on anything using QuantityLabel (100+ use cases), covering the requested cases like solar charger and PV inverter yields. There are still couple of dozen places, which call the old Units.getDisplayText() method, but a simple method call is more light-weight so would maybe ony add hysteresis to those per need. I still need to review the related use cases (e.g. VeBus alarm group, three-phase IO table, VeBus kWh counters, meteo page, inverter page, inverter charger page, EV charger list) and check if they have value labels that display large values
- Feels like 1000 would be better limit than 10000, but that will affect the resulting strings quite much and I am bit uneasy making such change without better understanding the previous design decisions regarding how labels should look
- The formatting code is quite complex so introduced autotests. Ran the same autotests against the old and new implementation to verify I am not causing regression. Noticed e.g. that previously the formatted values weren't rounded, i.e. 0.55 watts was shown as 0.5W (now fixed to show 0.6W). Also 0.4% and 0.4l were formatted as "0%" and "0ℓ" (now fixed to show 0.4% and "0.4ℓ")
- Previously the precision was fixed per type and means decimals after the decimal point. Changed the formatting to drop the decimals after the decimal point on larger numbers e.g. show "9321kWh" instead of "9321.31kWh"

The task has taken quite few days already, not sure medium priority task warrants this much efforts so close to 1.0 release. Hopefully inching now towards the finish line. See formatting examples in PR for more details, feedback is welcome.

jepefe, 2024-02-20:
@jpetrell 
This topic has come up for discussion on multiple occasions, but for some reason we never implemented such logic in gui v1. 
VictronConnect has a scaling logic but it is quite simple and it is only applied for history Wh <-> kWh. VictronConnect connects to standalone products, which, unlike systems, are less likely to go to those high values.

This is the conclusion of the last time we discussed the scaling logic:

From here limit the total digit count to a maximum of four, with no more than three digits allocated for the whole number component. Any remaining digit capacity is designated for decimal places.

Progression example:

```
9999Wh             -> 9999Wh // Base unit, no decimals.
10,999Wh            -> 10.99kWh
99,999Wh            -> 99.99kWh
999,999Wh           -> 999.9kWh
1,999,999Wh         -> 1.999MWh
99,999,999Wh        -> 99.99MWh
999,999,999Wh       -> 999.9MWh
1,210,999,999Wh     -> 1.210GWh
99,999,999,999Wh    -> 99.99GWh
999,999,999,999Wh   -> 999.9GWh
```

In my opinion this logic is quite straight forward to implement, and should work nicely for power and energy. 

Be aware that this or any other unit scaling will need some hysteresis to avoid bouncing between units. 

Same kind of logic can be applied to other measurements. Number of total digits and decimals might depend on what we are measuring.

jpetrell, 2024-02-20:
Thanks! I can align the behavior.

> 99,999,999Wh        -> 99.99MWh

Shouldn't we implement rounding? 99.99MWh feels misleading, when the actual value is closer to 100.0MWh.

> 9999Wh             -> 9999Wh
> 10,999Wh            -> 10.99kWh
> 1,999,999Wh         -> 1.999MWh

Feels odd that the jump from Wh to kWh has 10000 limit, but jump from kWh to MWh and MWh to GWh has limit 1000? Why the asymmetry?

jepefe, 2024-02-20:
> Shouldn't we implement rounding? 99.99MWh feels misleading, when the actual value is closer to 100.0MWh.

Rounding might be ok but which logic shall we follow?

```
99.99MWh      -> 100.0MWh // You are rounding 10kWh up which depending the context can be a lot, but ok.
99.96MWh      -> 100.0MWh // You are rounding 40kWh up which is definitively a lot of energy
```
I think that at least for power and energy, rounding is not needed.

> Feels odd that the jump from Wh to kWh has 10000 limit, but jump from kWh to MWh and MWh to GWh has limit 1000? Why the asymmetry?

Is all about making the logic simple. I added a clarification in bold, which was not in my original post.
- Maximum 4 digits
- Maximum 3 digits for whole part, rest for decimals. **This only if there are decimals, otherwise 4 digits rule applies.**

Units are not jumping by 10000 or 1000 limit but by the number of displayed digits. 

Main problem we are facing here is fitting big numbers in the UI and in second place the readability of them. This solution addresses both in a quite simple way.

Anyway, this is just a suggestion based on various discussions in the past. Perhaps we can find a better solution.

jepefe, 2024-02-20:
Hi again @jpetrell 

After some extra thinking, I noticed that we can solve the asymmetry by limiting the number of decimals to two.
Switching early to kWh is something that I'd really try to avoid. As a user with a domestic installation I do prefer to see W as much as possible and according to one of above Kevin comments, I'm not the only one.

Two decimals limit:

```
9999Wh              -> 9999Wh 
10,999Wh            -> 10.99kWh
99,999Wh            -> 99.99kWh
999,999Wh           -> 999.9kWh
1,000,000Wh         -> 1000kWh
99,999,999Wh        -> 99.99MWh
999,999,999Wh       -> 9999MWh
1,210,999,999Wh     -> 1.210GWh
99,999,999,999Wh    -> 99.99GWh
999,999,999,999Wh   -> 999.9GWh
```

jpetrell, 2024-02-20:
> Rounding might be ok but which logic shall we follow?

I would expect rounding used in e.g. scientific notation. Flooring feels conceptually wrong to me, but I am not a domain expert and if there is some inherent cost (to the users) in reporting rounded up figures, and the underestimation is somehow seen more beneficial can do that also?

> 99.96MWh      -> 100.0MWh // You are rounding 40kWh up which is definitively a lot of energy

Could also argue that 60kWh lost when flooring the value is a lot of energy? If you are dealing with megawatt-hours is 40kWh (i.e. 0.04% of 99.96MWh) really big? With the limited precision we are not showing the whole value, but partial presentation (which is also wanted since the measurement inaccuracies get accumulated in these figures). The issue is more notable in units that are shown in rough accuracy (e.g. displaying 1.9% as "1%" or -3.9° as "-3°").

> Is all about making the logic simple.

Not sure if that is more simple, but also it is not biggie. I guess there is not a big readability difference between 9999Wh vs. 9.999kWh. Could even consider the omission of decimal point beneficial, but the point anyway comes back haunting in larger scales (e.g. 1.209MWh).

jepefe, 2024-02-20:
>I would expect rounding used in e.g. scientific notation. Flooring feels conceptually wrong to me, but I am not a domain expert and if there is some inherent cost (to the users) in reporting rounded up figures, and the underestimation is somehow seen more beneficial can do that also?

It was a misunderstanding from my side. You meant to round up from the Wh (99,999,999Wh), while I was looking at rounding from the MWh (99.99MWh). Then yes, rounding should be applied. 

>Not sure if that is more simple, but also it is not biggie. I guess there is not a big readability difference between 9999Wh vs. 9.999kWh. Could even consider the omission of decimal point beneficial, but the point anyway comes back haunting in larger scales (e.g. 1.209MWh).

Check my previous comment. What do you think?

jpetrell, 2024-02-20:
> Check my previous comment. What do you think?

So to verify that I read that right you mean that we show e.g. 1234MWh instead of 1.234GWh? Looks good in that it makes all the scale ranges behave symmetrically, though now thinking more either way is really fine for me, would pick whichever people find more readable, e.g.  @kwindrem was asking for separators and having the decimal point in 1.234GWh kind of works like one.

I have now updated the PR to match your earlier example.

> As a user with a domestic installation I do prefer to see W as much as possible and according to one of above Kevin comments, I'm not the only one.

:+1:

jepefe, 2024-02-20:
> @kwindrem was asking for separators and having the decimal point in 1.234GWh kind of works like one.
This is a good point.

jepefe, 2024-02-20:
The logic I suggested is valid for power and energy. Voltage and current I think Matthijs already replied.

PR #745 (2024-02-28): Support giga, mega and tera ranges in value formatting
Fixes #620.
commit f59c5511 (2024-02-28): Support giga, mega and tera ranges in value formatting

Fixes #620.
closed 2024-02-28 by jpetrell