victronenergy/gui-v2 #634: Allow non-bundled gui-v2 builds

chriadam, 2023-12-22:
Currently, the QML files which define the gui-v2 UI are bundled into the executable as a resource.
In order to allow on-device editing of the UI files, we need to build and deploy gui-v2 differently (i.e. deploy the QML files as an installed module, and gui-v2 will load these files at runtime).

Most likely it will require changes to the CMakeLists.txt, e.g. the qt_qml_add_module() will need to be specified as a PLUGIN_TARGET with a specified OUTPUT_DIRECTORY etc.

This will have ramifications for the way the gui-v2 build artifacts are packaged and pulled into the firmware image, also.

kwindrem, 2023-12-22:
This is promising for those of us that modify the GUI. What impact will this have on the remote GUI that currently uses a wasm image?

Also, what impact will this have on performance both at startup and once the system is up? I would not want to slow responsiveness significantly.

chriadam, 2024-01-02:
For the WebAssembly one, everything has to be bundled into the executable blob anyway, so we will need to retain the current build setup for that target.

Regarding performance, we benchmarked extensively and using run-time loading of QML types does vastly slow down startup (due to requiring disk load + compile per-type on first resolution).  However, since this cost is primarily "hidden" by the splash screen, it isn't too prohibitive.

kwindrem, 2024-01-02:
If this won't work on the wasm build, then it's useless for third-party modifications. One third party mod could be handled by a recompile however combining multiple mods means combining the modified GUI files prior to compilation.

chriadam, 2024-01-02:
There might be a few different topics which are being conflated here.

1) unbundling the files and installing them to disk, and loading them at runtime, means that the user can directly modify the .qml we ship and see those changes (on the CerboGX hardware display).

2) but this is not the mechanism we intend to implement for "supported third-party modifications" which have been discussed elsewhere (e.g. adding new options to the settings page, etc) for which we have a few different options (e.g. load a .qml file at runtime either from a local or network source, or e.g. display a webview which allows arbitrary web content to be loaded, or whatever).

That is, this github issue is about allowing hobbyists to modify the files on device and see the changes on their CerboGX display.

If they ALSO want to see their changes appear in the WebAssembly version, they will need to install the sdk and create a new build of the wasm blob which they will then need to put under /var/www/venus/gui-beta/ and reboot the device so that the nginx will serve the new blob.  Yes, as you mention, if there are multiple mods, they will have to apply those in order, and build.