victronenergy/gui-v2 #636: Battery percentage is unnecessarily elided in Brief and Overview pages

blammit, 2023-12-22:
I'm seeing this when the battery charge is 100%. Noticed it with the browser/wasm, so it's possible it only happens in 7" screen size:

[image]

[image]

chriadam, 2024-01-02:
Ouch, thanks.

Serj has asked that we investigate the possibility of generating fixed-width bitmaps / pre-rendered images for each of the (horizontally-centered) digits rendered in the required font + font colour as specified by the theme (and font pixel size required for every label).

I was thinking it might be implemented via a QQuickImageProvider using query parameters to define font / pixelSize / color, which can render on first access then populate a cache...?

chriadam, 2024-01-12:
See #588 for the "proper" fix path, but for now I've created temporary fix in #659.

chriadam, 2024-01-31:
Bea fixed this properly in https://github.com/victronenergy/gui-v2/pull/671
Closing as done.

closed 2024-01-31 by chriadam