victronenergy/gui-v2 #639: Electron/Ants animations on the Overview page cause high CPU usage

chriadam, 2024-01-02:
Currently, we construct an interpolator for each WidgetConnector path on the Overview page.
Then, for each WidgetConnector path, we animate a "progress" property.
Each electron on the path listens to the `progressChanged` signal and then asks the interpolator for the appropriate x/y/rotation values given the progress value (adjusted for the electron's index).

Most of this happens in QML and so the calculations are slow, and cause a variety of change signals / wasted side effects.
Since the animation is consistently performed when the Overview page is visible, this causes very high CPU usage.

We have a few different options to try:

--

1) move most of the math to a C++ class which pre-calculates the value at every progress interval on the interpolator spline, and populates a hash table with this information for each electron (i.e. `QHash<pathInterpolator*, QHash<int electronIndex, qreal progress> >`).  Each electron merely looks up the appropriate value from this hashtable at each animation step.  Advantages: most closely matches the current implementation, and may reduce the amount of work done in QML expressions significantly.  Disadvantages: we'd still be performing an extremely large number of x/y/rotation property updates in the GUI thread (and triggering all of the changes signals etc), so I don't expect huge perf improvement.

2) since each WidgetConnector path is quite simple, we should be able to split each one into 5 sections (horizontal section, first bend, vertical section, second bend, horizontal section).  Each section defines x / y / rotation changes across their length.  Thus, we should be able to calculate just the start and end values for these three properties and use XAnimator/YAnimator/RotationAnimator to animate the values within each section.  Advantages: should greatly reduce CPU usage, as we'd be doing effectively (nbrSections * nbrProperties == 5 * 3) QQuickItem property writes per electron per animated path, rather than (nbrProperties * nbrAnimationTicks) QQuickItem property writes per electron per animated path.  Disadvantages: might be more difficult to calculate the start/end positions (and duration, i.e. based on pixel-velocity) than I predict.

3) we could just use a different method of animation altogether (e.g. could approximate the current design with a moving zebra-pattern texture opacity masked to the connector).  Advantages: performance would be much better.  Disadvantages: would look significantly worse.

4) something else?

--

I will try (2) first.

mr-manuel, 2024-01-02:
What if you overlay an animated SVG if the electrons are moving?

chriadam, 2024-01-02:
> What if you overlay an animated SVG if the electrons are moving?

Possible!  We could use a custom QSvgRenderer-backed item to draw the electrons.  The problem is that we would have to draw them with QPainter, I believe (unless AnimatedImage supports animated SVGs as well as GIFs - but I doubt it...).  QPainter is software rendering on the CPU so it might prove to be somewhat expensive, unfortunately.  But this is definitely worth trying if (2) doesn't work.  Thank you for the suggestion!

DanielMcInnes, 2024-02-07:
Consider adding a 'lightweight animations' setting to allow decent performance on heavily loaded systems. Some heavily loaded systems are running out of cpu when running gui-v1.

jpetrell, 2024-02-07:
I fear most people suffering from the issue would not realize there is such a setting and successfully locate it in the current settings hierarchy. Unless the high CPU issue can be fixed for the most demanding use cases would maybe try to detect the issue somehow automatically, i.e. after some threshold (amount of connections, CPU limit, etc.) the costly animations are replaced by some light-weight design, either less taxing animations or even stationary graphical states to indicate flows and directions.

mpvader, 2024-02-07:
Hey both, the plan we agreed on was slightly different, so lets not now get started on discussing such lightweight animations yet:

1. Chris implements his ideas to improve the implementation of those animation.
2. We compare CPU usage on a heavily loaded system. Comparison is between gui-v1 on overview page versus gui-v2-most-heavy-page
3. If CPU usage of gui-v2 is comparable to gui-v1 then no need to get into any lightweight option; certainly not now or per v1.0.0. 

The lightweight animations will only be part of v1.0.0 if we can't reduce gui-v2 CPU within a reasonable implementation time

jpetrell, 2024-03-04:
Here is a PR that should reduce overall CPU usage by ~10-20%:

https://github.com/victronenergy/gui-v2/pull/827

- With 20 electrons the original CPU 60% usage reduced to 50% i.e. roughly 17% improvement
- With 8 electrons the original CPU 51% usage reduced to 46% i.e. roughly 10% improvement

Took the measurements on Cerbo GX.

Seems the overall performance is mostly bound to the number bitmaps being drawn at once. As a next step would be nice to try rewriting the path animation fully using shaders. Implementing such a shader would be a breeze if the Qt Quick Particles supported custom paths (=define emitter emitting bitmap particles at regular intervals, particles moving and rotating along the path attractor at a constant progress velocity and predefined fade timeout) or even by defining custom vertex shaders for the particles, unfortunately neither approaches are currently supported by Qt (e.g. CustomParticle element was dropped in Qt 6.0) so the implementation would require a bit more effort.

jpetrell, 2024-03-07:
> a PR that should reduce overall CPU usage by ~10-20%:
> With 20 electrons the original CPU 60% usage reduced to 50% i.e. roughly 17% improvement

This is not enough, need to get the Overview page CPU consumption below 25%, which requires a bigger rewrite. Though I am not sure we can do electron animations on Cerbo at 25% CPU. Even th environment gauges on the Levels page run at 31% CPU in 60fps with little animation. The Brief page can do 10% CPU, but runs at only 18-20fps.

One issue is the high 60fps frame rate. The Overview page can only draw 34-36fps on Cerbo GX. If we improve the performance the extra bandwidth will just go into increasing the fps until it caps at 60fps.

The frame rate already has huge range of variation, only static pages like settings can do 60fps (idle and scrolling) so would suggest to limit the frame rate to 30fps on Cerbo (Ekrano seems to handle 60fps better). Experience-wise it is better to do solid 30fps than choppy 60fps.

jpetrell, 2024-03-07:
Did some measurements on Cerbo CX. Without limiting the global frame rate reaching the 25% CPU limit will be difficult. Animation optimizations will only give us small, incremental improvements to the ~33-40% baseline CPU consumption we see with the static page layouts.

```
Brief page                        CPU 15% 18fps
Notifications page (empty)        CPU 33% 60fps
Main settings page                CPU 38% 60fps

Levels page
3 Tank gauges                     CPU 40% 60fps
4 Environment gauges              CPU 47% 60fps

Overview page
20 Electrons          30 Bubbles  CPU 60% 33-37fps
20 Animator electrons 30 Bubbles  CPU 46% 35-37fps
 0 Electrons          30 Bubbles  CPU 44% 35-37fps
20 Animator electrons  0 Bubbles  CPU 40% 44-50fps
 0 Electrons,          1 Bubbles  CPU 39% 46-50fps
 1 Animator electron   0 Bubbles  CPU 42% 50-60fps
 0 Electrons           0 Bubbles  CPU 43% 57-60fps
```

chriadam, 2024-03-07:
Please get QML analyzer traces and find out what is taking the CPU load.  (Please also ensure that you have stopped gui-v1 during your testing, and that it's not running in the background, with `svc -d /service/start-gui` etc.)

chriadam, 2024-03-07:
Oh, for completeness, the "3%" value I mention is the value that `top` reported for the gui-v2 process only, not total CPU usage of entire device.

jpetrell, 2024-03-07:
> But 40% CPU on a static (Levels) page is simply insane, there must be errant animations or something going on there now.

Oh true, I had added local FpsMonitor component in there, which plays a constant animation to measure the frame rate. Without it the CPU usage on static Tanks and Settings pages goes to only few percentages or less. So the 33-40% baseline CPU consumption mentioned earlier is from anything that plays even the tiniest animations, not the static page layouts sorry. FpsMonitor updates the frame rate label once a second, but constantly animates one NumberAnimation.

chriadam, 2024-03-10:
Wait, having a NumberAnimation always drives the CPU usage up to 45%?  Even if the property the NumberAnimation is driving doesn't affect rendering or something?  Are we CERTAIN that the animation step isn't just "as fast as possible" rather than "synced to vsync"?  What I'm wondering is, for the overview page at least, would it be better to write our own 32 ms fine-grained timer to trigger the electron animation steps manually?

jpetrell, 2024-03-11:
There are three different ways to limit the effective frame rate:

1. Limit the refresh rate on the display driver level.
- No need for higher display refresh rate than the system GUI
- Limiting physical display refresh rate is hard
- Can connect Cerbo to any HDMI display, not all displays support 30fps

2.) Limit the maximum frame rate of Qt scene graph.
- Qt doesn't make it particularly easy to reduce the frame rate
- Still might be nice to support 60fps scrolling but 30fps idle animations
- May cause tearing and stutter if vsync is disabled

3.) Use FrameAnimation to limit the frame rate of individual animations.
- Relatively straight-forward to implement
- Only provides performance benefits if all independent animation and label updates use the same sub-refresh rate heartbeat
- FrameAnimation is bound to the main-thread, performance-wise Animators that run on different thread might have been better

Discussed with @mansr that limiting display refresh rate i.e. option 1.) does not seem like a good idea. It might be more feasible on devices like Ekrano with fixed display configuration, but you can connect Cerbo GX to any HDMI display, and hacking with driver display configs is asking for trouble.

I tried 3.) limiting the animation frame rates with FrameAnimation. FrameAnimation does help, but alone won't get us below the target 25% figure:

```
20 Electrons  0 Bubbles 50fps CPU 45%
20 Electrons  0 Bubbles 30fps CPU 41%
20 Electrons  0 Bubbles 20fps CPU 36%
20 Electrons  0 Bubbles 15fps CPU 33%
20 Electrons  0 Bubbles 10fps CPU 30%
20 Electrons  30 Bubbles 20fps CPU 37-40%
20 Electrons  30 Bubbles 15fps CPU 32-35%
```

Tiny widget connector animations look smooth on 30fps and still ok with 15fps, but the battery bubble animation starts to be visibly choppy already in 15-20fps range.

The resulting code becomes bit convoluted when using the FrameAnimation, but maybe that is the price we have to pay for optimizations:
https://github.com/jpetrell/gui-v2/commits/overview-electron-optimization-global/

If we would limit the animation frame rate *and* rewrote both the battery and widget connector animation we should at least get below 30%.

I wonder if limiting the global Qt frame rate i.e. option 2.) would yield better results (or same as FrameAnimation), but so far (unless we fork Qt) seems the Qt Quick doesn't allow limiting the maximum frame rate below the screen refresh rate.

Some investigation results:
- Tested that increasing animation timing interval from 16ms to 32ms with `QUnifiedTimer::instance()->setTimingInterval()` just doubles the speed of the animations.
- [Driving animations chapter]( https://doc.qt.io/qt-6/qtquick-visualcanvas-scenegraph.html#driving-animations) in Qt Quick Scene Graph documentation says that "Qt Quick is prepared for other refresh rates as well: the rate is queried from the QScreen and the platform.". 
- Altering the animation refresh interval in Qt Quick would require defining a custom QAnimationDriver, but QQuickWindow seems to be hard-coded to QSGAnimationDriver
- QSGAnimationDriver calculates the animation interval from the screen refresh rate that is queried from the eglfs platform plugin, which subsequently reads it from the OS framebuffer device configs

I either could try a.) alter the screen refresh rate reported to Qt by modifying the eglfs platform plugin or b.) alter the animation refresh rate by modifying QtDeclarative library / QSGAnimationDriver to see if anything breaks and if we get different performance results compared to the FrameAnimation.

jpetrell, 2024-03-11:
> Are we CERTAIN that the animation step isn't just "as fast as possible" rather than "synced to vsync"

Yes. Qt Animations trigger updates at 60fps (if the app performance permits).

chriadam, 2024-03-11:
> > Are we CERTAIN that the animation step isn't just "as fast as possible" rather than "synced to vsync"
> 
> Yes. Qt Animations trigger updates at 60fps (if the app performance permits).

It's mind blowing that simply writing to a single int property once every 16 ms could cause 38% CPU usage...

jpetrell, 2024-03-11:
> It's mind blowing that simply writing to a single int property once every 16 ms could cause 38% CPU usage...

Well that only happens if the vsync is enabled so I guess more than int property writes are happening.

jpetrell, 2024-03-11:
> I would really like to see QML analyzer traces from the NumberAnimation on single int property test case, from device.

I can take such traces tomorrow, though I suspect QML analyzer won't show much as not much is really happening on the QML execution side, just some small, negligible JavaScript expression evaluated on every frame.

jpetrell, 2024-03-11:
> https://github.com/jpetrell/gui-v2/commits/overview-electron-optimization-global

Finished FrameAnimator implementation ^^. Unfortunately replacing BatteryWidget bubble Animators fully with imperative updates (tried both JavaScript and C++) is clearly more taxing to CPU, so while reducing the animation frame rate reduces the overall CPU usage the benefits are nullified by the slower animation paths.

20 electrons, 30 bubbles:
```
50fps CPU 60%
20fps CPU 48%
10fps CPU 43%
1fps  CPU 15%
0fps  CPU 1%
```

jpetrell, 2024-03-13:
Serj provided a simpler charging animation. With the optimizations so far and the new charging animation the Overview page performance on Cerbo GX goes down from 58% to 44% (20 electrons): https://github.com/victronenergy/gui-v2/pull/827 . Tested that rewriting the electron animations with animators could potentially get us further down to 42%, but not sure rewrite makes sense for such a minor improvement (also if I have learned something doing this task is that many of the performance benefits tend to melt away when moving from a naive test to a feature complete solution). Remember that animating any small item at 60fps is enough to cause 40% of CPU usage on Cerbo GX.

Got a display patch from @mansr to test how much reducing the refresh rate would help with the CPU load, which ended up reducing the frame rate from 60fps to 36fps (reported by the QML animations). Lowering the frame rate reduced the CPU usage from 57% to 49% on the current main branch. Combining the lower frame rate with the existing optimizations got the CPU usage all the way down to 33% (with 30fps could get us blow the 30% mark). Though these figures were taken with GX Touch 50, and the earlier figures have been measured using a larger GX Touch 70 display so not sure how comparable these are.

mpvader, 2024-03-13:
As discussed today:

- improvements done now will go into master
- further work on this, including working on having Qt operate at reduced frame rate, is paused.
- for CPU overloaded systems, we're making an automatically engaged fallback that has no continuous animations at all. Not in the battery and not electrons. There will be a new issue for that. Implementing that is just a few hours work; and first will be a design.

Later we'll continue on this, most likely by reducing frame rate since thats the most effective change. Reducing frame rate requires patching Qt.

jpetrell, 2024-03-14:
Filed issues https://bugreports.qt.io/browse/QTBUG-123320 tand https://bugreports.qt.io/browse/QTBUG-123317 to Qt about the animations causing high CPU usage and Qt lacking an option to decouple the Qt Quick drawing from the screen refresh rate.

jpetrell, 2024-03-20:
The incremental performance improvement was merged. Marking the task as blocked, the remaining work of replacing electricity direction animations with static indications during heavy loads is pending both design and middleware (to indicate when the system is under heavy load).

Discussed with Serj that the battery animation doesn't need a static charging indicator as the battery box already clearly shows "Charging" state as a label, but he will think more about how replace the electricity direction animation with static indicators.

I am a bit concerned how the overload detection will be done in that I hope we can avoid cases where the system constantly oscillates between the animated and static states. If user sees the electricity animations suddenly pausing on the screen it may look like the electricity stopped flowing even if the static indicators otherwise display the directionality clearly.

jpetrell, 2024-03-27:
Found the reason for the high CPU usage. The other day I accidentally resized the gui-v2 screen on desktop and to my surprise saw the Qt virtual keyboard peeking below the normal view area. The virtual keyboard is drawn on every frame, even when it has been moved out of the view. On Cerbo, hiding the unused keyboard reduces the idle animation CPU usage from 40% to 24%.

See PR https://github.com/victronenergy/gui-v2/pull/904

CPU figures measured on Cerbo:
```
Keyboard            Visible     Hidden
Minimal animation   CPU 40%     CPU 24%
Overview page       CPU 44%     CPU 34%
While scrolling     CPU 58%     CPU 45%
```
On Ekrano minimal animation CPU usage drops from 17% to 9%.

Keyboard wasn't visible before, because it had been moved out of the view. That wasn't enough, because Qt Quick scene graph doesn't do any viewport culling, but tries to draw all the items marked as visible on the screen (AFAIK with the low-complexity traditional 2D UIs the culling often ends up being more costly than any of the gained savings). Virtual keyboard is a visually complex widget with dozens of labels, icons and backgrounds, often even more complex that is shown on the regular UI page.

mpvader, 2024-03-27:
Nice find!

chriadam, 2024-04-01:
Nice find!  My bad, I thought I hid all offscreen things properly.

jpetrell, 2024-04-02:
As discussed filed a separate issue https://github.com/victronenergy/gui-v2/issues/914 for pausing animations on high CPU usage.

Closing this one.

closed 2024-04-02 by jpetrell