victronenergy/gui-v2 #650: Issues with analog tank inputs (13,208 gallons + 3rd tank input issues)

mpvader, 2024-01-10:
For starters, @ReinvdZee will see to reproduce it.

https://community.victronenergy.com/questions/249509/venus-os-v32034-available-for-testing.html

[image]

ReinvdZee, 2024-01-17:
**Tested on Cerbo GX running V3.20\~37**
I did some testing and found the following issues with the tanks:

### 1. Switching from GUI v1 to Beta --> Tank capacity off by 1
I started off with two tanks configured in GUI v1:

1. Fuel tank of 100gal
2. Fresh water tank of 200gal
I connected some resistors to the tank input of the GX Tank 140 to end up with the following:
[image]

After switching to the beta GUI, the tank capacity levels are decreased by 1:

[image]

The setup section shows the correct value. However, the edit menu shows the value off by 1 again.
[image]

### 2. Incorrect capacity after changing
Next, I changed the capacity of the fresh water tank from 199 (or 200) to 205gal:
[image]

The new capacity setting seems to be saved properly:
[image]

However, switching to the "Levels" view yields a capacity of 54155 gal:
[image]

**54155gal equals 205m3, issue in capacity conversion between units? dbus-spy also shows a capacity of 205m3**

I am also unable to edit the capacity level now, I tried to change the capacity to 54000gal. dbus-spy still shows a capacity of 205m3. Changing the capacity all the way back to 200gal, does have effect, it modifies the capacity to 52834gal (200m3).

### 3. Improper unit conversion
I took the following steps in changing the capacity and units of the tank:

1. Start with 1 cubic meter
2. Convert to liters, yields 1 liter
3. Set to 500 liter
4. Convert to cubic meters, yields 500m3
Under the assumption that the capacity should be converted to the newly entered unit (500l --> 0.5m3), this behavior is incorrect. 

### 4. All tanks get the same unit
Next, I tried to change the volume unit of the fuel tank to m3, while keeping the unit of the fresh water tank set to gallons. However, both tanks changed to m3 units. 

[image]

### 5. Tank fill level incorrect in "Levels" view
WIth the tanks having a capacity of 1m3, and the sensor indicating fill levels of 75% and 42%, the "Level" view shows a remaining volume of 0/1m3:
[image]

### 6. Tank type change not shown
I changed the type of the "Fuel" tank to "Diesel", but the tank still shows up as "Fuel" tank in "Levels". Refreshing the webpage (WASM gui), udpates the view correctly. On the device, a reboot (or probably a GUI restart) was necessary for the change to become apparent.

blammit, 2024-01-26:
Thanks for the thorough testing @ReinvdZee. https://github.com/victronenergy/gui-v2/pull/685 addresses these issues, with the exception of "4. All tanks get the same unit". The current behavior of using the same unit for all tanks should be sufficient, as it reflects the gui-v1 behavior and there are no settings to store per-tank units, but this could be added as a separate feature in the future if needed.

mpvader, 2024-01-26:
> The current behavior of using the same unit for all tanks should be sufficient

indeed that is sufficient. We don't want a setting to per tank store the unit (l, m3, gallon etc)

closed 2024-01-30 by blammit