victronenergy/gui-v2 #663: Make dialogs modal

blammit, 2024-01-16:
Don't allow dialogs to be dismissed by clicks outside of the dialog area. The result is ambiguous when a dialog only has "set" and "cancel" buttons, and all dialogs have some sort of close buttons anyway.

This should also help avoid this crash bug on wasm when clicking outside of the dialog: (reported in https://github.com/victronenergy/gui-v2/pull/654)

[image]

DanielMcInnes, 2024-01-22:
I filed a bug report for this, https://bugreports.qt.io/browse/QTBUG-121382 .
I notice that you can stop this crashing by either:
1: removing the `modal: true` part from SolarDailyHistoryDialog (this changes the behaviour, clicking outside the dialog box, but inside the chart area opens a new dialog showing a different day's data)
2: removing the `exit: opacity animation{}`. Looks a bit jarring, but retains the existing behaviour

closed 2024-01-23 by DanielMcInnes

blammit, 2024-01-24:
Made https://github.com/victronenergy/gui-v2/issues/676 to keep the "make dialogs modal" part of the task.