victronenergy/gui-v2 #669: Change the order of the cards in the control section

mpvader, 2024-01-18:
1. ESS
2. Generator
3. Multi

Rest comes later. right now ESS is by default half hidden; while its the most used one (in systems that have it). See below for how it looks now.

As a bonus, (perhaps let us know how much work it will be to implement @blammit ?), it would be great if the gui can count how often certain controls are used, and shows the card with the most used controls first.

[image]

blammit, 2024-01-19:
> As a bonus, (perhaps let us know how much work it will be to implement @blammit ?), it would be great if the gui can count how often certain controls are used, and shows the card with the most used controls first. And there also the time needs to be taken into account. If you need more about that time idea, ask Jesus. Perhaps best to put this into a separate issue; for sure this is a phase-2 thing; not for the first release.

We could calculate usage based on mouse interaction, but It depends what you mean by "used" -- what if a user likes to view a particular card often, but the user doesn't actually need to interact with it? How would we determine whether it's "in use"? Would it be counted based on the time that the card is fully in view? But then how do you offset that against the number of times the user interacts with another card? Hopefully it would not be too confusing to the user if the order of the cards changed suddenly.

With regards to counting/time, if this needs to be saved across sessions, then some setting under `com.victronenergy.settings` needs to be added -- probably a map of `{ card_name: seconds_used }`.

mpvader, 2024-01-19:
> but It depends what you mean by "used"

Interact with it: change one of the controls on the card.

with time I meant time of day. Some controls are used more often at different times of the day.

but this is not for forst release (!).

blammit, 2024-01-22:
> > but It depends what you mean by "used"
> 
> Interact with it: change one of the controls on the card.
> 
> with time I meant time of day. Some controls are used more often at different times of the day.
> 
> but this is not for first release (!).

Okay, thanks, I've made https://github.com/victronenergy/gui-v2/issues/672 for that.

kwindrem, 2024-01-22:
I don't think this is a good idea.

A great deal of user familiarity with a user interface is based on a known path so some function or parameter. Think how often we struggle when Apple or Microsoft moves a setting or menu option.

closed 2024-01-25 by blammit