victronenergy/gui-v2 #672: Order the control cards depending on usage

blammit, 2024-01-22:
As discussed in https://github.com/victronenergy/gui-v2/issues/669.

"It would be great if the gui can count how often certain controls are used, and shows the card with the most used controls first. And there also the time needs to be taken into account. If you need more about that time idea, ask Jesus. Perhaps best to put this into a separate issue; for sure this is a phase-2 thing; not for the first release."

kwindrem, 2024-01-22:
I don't think this is a good idea.

A great deal of user familiarity with a user interface is based on a known path so some function or parameter. Think how often we struggle when Apple or Microsoft moves a setting or menu option.

chriadam, 2024-01-31:
Not for 1.0.0, marking as low_prio.