victronenergy/gui-v2 #674: Test gui-v1 vs gui-v2

mpvader, 2024-01-22:
Rein has been, and will be further, checking and testing gui-v2 versus gui-v1.

Results are here: https://docs.google.com/spreadsheets/d/1xMHrUgUGXaVYIKtAQ3cdD3pbpgExw1qBrLF4bKw7zkk/edit#gid=0

Next step is that Rein will do the tests again, but then using gui-v2 in a browser instead of on an Ekrano or Cerbo.

ReinvdZee, 2024-01-25:
I've been doing some testing on the GUI-v2 in my browser. See the "Browser" tab in the shared excel sheet. 

Note that these tests have been performed with my cerbo running v3.20\~40, whereas the tests in the "Local" (GUI tested locally) are performed on a cerbo/ekrano running v3.20\~35. 

**Some other, possibly unrelated issue:**
Now and then, I am unable to run the GUI in the browser. My browser hangs on the "Connecting" state. The browser complains about not being able to connect to port 9001. I discovered that when this happens, there seems to be an iptables rule preventing my connection: 

`REJECT     tcp  --  anywhere             anywhere             tcp dpt:9001 reject-with icmp-port-unreachable` 

After dropping this rule and refreshing the browser, the GUI is loaded properly. 

This iptables rule seems to reappear when disabling MQTT on lan (Plain-text). This might make sense, but the rule is not deleted when re-enabling MQTT.

jpetrell, 2024-02-13:
>> Results are here: https://docs.google.com/spreadsheets/d/1xMHrUgUGXaVYIKtAQ3cdD3pbpgExw1qBrLF4bKw7zkk/edit#gid=0
> Toast about disabling VE.Bus shows up after enabling DVCC and not after disabling DVCC.

Fixed by #733.

> Cannot switch off accesspoint. Warning showing, but AP not switching off.

Fixed by #727.

jpetrell, 2024-02-13:
There are quite many findings reported in Local and Browser tabs that, if they still reproduce, would be good to get reported here in GitHub. Also, earlier e.g. the broken keyboard broke so many settings that it is potentially masking other issues underneath.

ReinvdZee, 2024-02-14:
@jpetrell, on V3.30~4, the keyboard on the WASM build is not functional. On the device, the OSK works properly.

ReinvdZee, 2024-02-14:
I'm currently retesting some issues in Gui-v2. I'll create issues in `gui-v2` with the `Version 1.0.0` milestone set.

mpvader, 2024-02-14:
Hey both @ReinvdZee & @jpetrell 

Wrt the broken keyboard in WASM: that is fixed in the latest WASM build - gui-v2 v0.2.10, and already listed on the todo for inclusion in the next build of Venus OS. And has also mentioned in many other issues. Lets stop discussing that :-)

ReinvdZee, 2024-02-14:
@mpvader,

Alright, I was not aware the WASM OSK was already under discussion, thanks!

ReinvdZee, 2024-03-26:
Closing this issue, gui-v2 issues will be reported in the gui-v2 repository directly.