victronenergy/gui-v2 #677: Make text input work in wasm

blammit, 2024-01-24:
Text input is broken in wasm due to https://bugreports.qt.io/browse/QTBUG-116557 (TextField in StackView does not receive key events). This was fixed in Qt 6.5.4, but that is a commercial release so it is not compatible.

Some options:

* Fix it within the current Qt 6.5.2 build by porting the relevant patches (as per the Qt bug)
* Drop uses of StackView altogether (probably not reasonable, but depends on what happens with https://github.com/victronenergy/gui-v2/issues/538)

chriadam, 2024-01-25:
Note: this might already be fixed in the official builds (@dirkjanfaber might already be patching Qt 6.5.2 used in the github action builds) - see discussion on slack and comments on issue #441 also.  Please confirm with him!

dirkjanfaber, 2024-01-25:
In the build action, I am not building Qt, but using the default 6.5.2 install (which I do patch, but that is mainly adjusting some paths and permissions). In my install I don't have the Qt source, so cannot use [this](https://code.qt.io/cgit/qt/qtbase.git/diff/?h=6.5&id=110286341a0a745aa505a10cd39eb56148e53d9f) patch there.

chriadam, 2024-01-31:
Ok, so am I correct to state that our goal is to upgrade to Qt 6.6.x for the wasm builds - we don't need it for cerbo gx builds since this bug only affects wasm. @dirkjanfaber 

@DanielMcInnes and @chriadam need to locally test the latest opensource Qt 6.6.x wasm build to ensure it works, before Dirk-Jan spends any effort here.

Highest priority.

DanielMcInnes, 2024-01-31:
I tested Qt6.6.1 wasm, the text input issue is fixed

DanielMcInnes, 2024-02-07:
Qt6.6.1 is only really necessary for wasm builds. If it is easy to move all builds to Qt6.6.1, do it. If it's hard, we can work around it

DanielMcInnes, 2024-02-07:
Let's not spend much effort on this yet before doing some proper testing on 6.6.1. If no obvious regressions, go ahead.

dirkjanfaber, 2024-02-09:
My [`faberd/qt-6.6.1-emscripten-3.1.37`](https://github.com/victronenergy/gui-v2/tree/faberd/qt-6.6.1-emscripten-3.1.37) branch builds with Qt6.6.6.1 and emscripten 3.1.37. To me everything seems to work alright (though mouse scrolling seems to be a bit slower). Text input does work on the webassembly builds.

@chriadam / @DanielMcInnes : can you check if this resolves all and merge if ok?

blammit, 2024-02-19:
Verified that this is fixed with the Qt+emscripten upgrade.

closed 2024-02-19 by blammit