victronenergy/gui-v2 #680: Brief tank levels for Black and Waste are inverted

mpvader, 2024-01-24:
The same data is shown correctly on the Tanks view.

Reported here: https://community.victronenergy.com/questions/249509/venus-os-v32034-available-for-testing.html?page=2&pageSize=50&sort=oldest

[image]

chriadam, 2024-01-31:
I believe this issue was fixed in https://github.com/victronenergy/gui-v2/pull/685 but we need to confirm.
If this cannot be reproduced with latest release, we can close as fixed.

gharshaw, 2024-02-21:
I just reinstalled with v3.30~6 and the waste and black are still inverted on the brief view

blammit, 2024-02-27:
@gharshaw could you share screenshots of the values in `Settings > Debug > Values > Tank > [tank number]`? Alternatively if you add victron.vrm (at) qinetic.com.au to the list of users in the VRM portal then we can investigate directly.

blammit, 2024-02-27:
@gharshaw Thanks! In the values, are there any "Capacity" and "Remaining" values?

gharshaw, 2024-02-27:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

blammit, 2024-03-12:
> @gharshaw could you share screenshots of the values in `Settings > Debug > Values > Tank > [tank number]`? Alternatively if you add victron.vrm (at) qinetic.com.au to the list of users in the VRM portal then we can investigate directly.

@gharshaw I'm not able to reproduce this locally. Would you be able to add us as a user to your installation on VRM portal so that we can test and verify directly? 

Black and Waste Water should be inverted in colors only, and not in the displayed percentage.

gharshaw, 2024-03-12:
[image]
[image]
Brief view is showing that Black and Gray Tank are 100% full. The tank view shows that they are 0% full or empty. 0% or empty is the correct state of the tanks. -

blammit, 2024-03-13:
Thanks! Will look further into this.

chriadam, 2024-03-19:
Hi @gharshaw I tried to connect to your installation via VRM today, but it looks like you don't have real-time data updates enabled.  Could you double check that Settings -> VRM Online Portal -> VRM Two Way Communication is enabled, please?

gharshaw, 2024-03-19:
@chriadam 

As you suspected. Two way is now enabled

chriadam, 2024-03-20:
@gharshaw thanks very much, have created PR with the fix now.

chriadam, 2024-03-25:
As per above comment, merged PR then set issue as blocked, low-priority.

gharshaw, 2024-03-28:
@chriadam
Appears to be fixed in v0.2.14 
[image]
:+1:

jpetrell, 2024-04-10:
What is still needed before the issue can be closed? Can the remaining bit filed as a separate issue?

chriadam, 2024-04-12:
Created issue #937 for the remaining bit, will close this task now.

commit 829783c3 (2024-03-22): Avoid calculating tank level on Brief Page gauges if possible

If the tank reports its own level, we don't need to calculate
the level.

Note that in some cases it seems like a tank will report
a different level to what we would have calculated
(i.e. we calculate level as being 100*(remaining/capacity)
whereas some black/waste water tanks report level as being
100*(1.0 - (remaining/capacity)).

Fixes #680
commit b27ec919 (2024-03-22): Avoid calculating tank level on Brief Page gauges if possible

If the tank reports its own level, we don't need to calculate
the level.

Note that in some cases it seems like a tank will report
a different level to what we would have calculated
(i.e. we calculate level as being 100*(remaining/capacity)
whereas some black/waste water tanks report level as being
100*(1.0 - (remaining/capacity)).

Fixes #680
PR #891 (2024-03-25): Avoid calculating tank level on Brief Page gauges if possible
Affects WasteWater and BlackWater tanks.

Fixes #680
commit 92295fbd (2024-03-25): Avoid calculating tank level on Brief Page gauges if possible

If the tank reports its own level, we don't need to calculate
the level.

Note that in some cases it seems like a tank will report
a different level to what we would have calculated
(i.e. we calculate level as being 100*(remaining/capacity)
whereas some black/waste water tanks report level as being
100*(1.0 - (remaining/capacity)).

Fixes #680
closed 2024-04-12 by chriadam