victronenergy/gui-v2 #681: Generator auto-start hours vs minutes mixup

mpvader, 2024-01-24:
reported here: https://community.victronenergy.com/questions/249509/venus-os-v32034-available-for-testing.html?page=2&pageSize=50&sort=oldest

[image]

chriadam, 2024-01-31:
Should be simple fix.  High priority.

jpetrell, 2024-02-06:
Debugged the UI side (GeneratorStartDialog + Utils.composeDuration()), AFAICS the hours and minutes are correctly turned into seconds that are subsequently passed to VeQuickItem.setValue() that has uid "{serviceUid}/ManualStartTimer".

DanielMcInnes, 2024-02-07:
This should be testable without a generator, it just opens a relay

jpetrell, 2024-02-07:
Tested with GFX Ekrano. The hour and minute values are not inverted and I can see that the GUI sets the correct ManualStartTimer value to the system using the dbus-spy, but occasionally something does seem to cause the generator to turn off too early. Setting the timed run to 1 minute lasted 1 minute and setting it to 2 minutes lasted 2 minutes, but when I set the timed run to 1 hour the generator got turned off (=heard the relay snap sound) after roughly 5 minutes. I am not sure what exactly turned off the relay prematurely, but looking at the QML code it is not likely coming from the GUI.

mpvader, 2024-02-28:
discussed today: @jpetrell will spent 3 more hours trying to reproduce it; if he can - look with Rein. If not, close it.

jpetrell, 2024-02-28:
Tested 1 minute, 1 hour & 2 minutes, 2 hour & 5 minutes. The hours and minutes are saved correctly and the timed run works as expected. Cannot reproduce. Might have been an older issue that has since been fixed. Closing.

closed 2024-02-28 by jpetrell