victronenergy/gui-v2 #682: Generator control card: enabling auto start doesn't work?

mpvader, 2024-01-24:
Reported by a user, see 2nd screenshot, and I can also reproduce this myself.

Note that (a) when doing the same (enabling autostart) from the Settings menu, all works fine. Clicking the toggle switch in gui-v2 (WASM) immediately makes for the toggle to switch in Remote Console (old gui) as well. and (b) disabling does work well.

The fault state:

[image]

The report by the user:
[image]

https://community.victronenergy.com/questions/249509/venus-os-v32034-available-for-testing.html?page=2&pageSize=50&sort=oldest

DanielMcInnes, 2024-02-07:
M.V can reproduce, so ask him if you can't

jpetrell, 2024-02-07:
Fix PR https://github.com/victronenergy/gui-v2/pull/719

mpvader, 2024-02-27:
This one can / should’ve been closed?

ReinvdZee, 2024-02-27:
@jpetrell, @mpvader,

I am still able to reproduce this issue on V3.30~7, running gui-v2 v0.2.10 (afaict this PR is merged in this version).

[image]

When controlling the autostart functionality from gui-v1, the control card in gui-v2 is not following properly:
[image]

The autostart switch on gui-v2 can be either on or off. dbus path `Settings/Generator0/AutoStartEnabled` correctly follows gui-v1.

jpetrell, 2024-02-28:
> I am still able to reproduce this issue

Earlier PR fixed enabling autostart from the generator control card, but after the PR the switch no longer updated if the autostart value changed elsewhere. Should be fixed now in https://github.com/victronenergy/gui-v2/pull/812.

PR #812 (2024-02-29): Don't break switch binding to the /AutoStartEnabled
If the autostart value changed elsewhere the switch on the generator card wasn't updated.

Fixes #682.
commit e77aed6a (2024-02-29): Don't break switch binding to the /AutoStartEnabled

Fixes #682.
closed 2024-02-29 by jpetrell