victronenergy/gui-v2 #687: gui-v2 crashed when opening generator device

mpvader, 2024-01-26:
https://www.youtube.com/watch?v=GW9gaJ5qhSg

blammit, 2024-01-31:
I think this is related to a VeQuickItem uid being invalid when the page is initially loaded, which should be fixed by 28b03d2888933ad0580f7828871c354729c516b0 and other recent veutil changes. However, it would be nice to test this on a real system if I knew the installation or configuration -- I haven't been able to reproduce the crash when trying some of the venus-docker simulations.

chriadam, 2024-01-31:
It would have only crashed with dbus backend since the Utils.normalizedSource() method (which I've removed in that change you mentioned) only stripped the dbus path prefix.  So trying to repro with a remote installation would never have triggered the issue.

I reproed on cerbogx and fixed.

chriadam, 2024-01-31:
High prio.

chriadam, 2024-02-01:
Can we get some more information about the product, e.g. the product id?
I assume the InteliLite is a type of generator / genset device?
Is it a Fischer Panda device, or something else?

Certainly, my PR should have resolved the crash due to any sort of transient invalid UID state, but I would like to be able to trace the specific codepath taken for this device - for which I need the product id etc.

mpvader, 2024-02-01:
Yes an Intelilite is a model generator start/stop controllers, its from Comap. Setting up a bench test for that is not so easy, I've asked the user that reported this to confirm that its now fixed.

For me its fine if you close meanwhile this issue; it will come back if there is still an issue.

[image]

mpvader, 2024-02-02:
Just got the confirmation, fixed indeed @chriadam

closed 2024-02-02 by mpvader