victronenergy/gui-v2 #700: After some timeout, return to Brief or Overview page

blammit, 2024-01-31:
After 10 minutes of inactivity, return to the Brief or Overview page, whichever one was viewed last. (As per Slack discussion)

Current proposal is to do this without adding a user option on whether this behavior should occur.

Implementation details:
gui-v2 can use the existing `com.victronenergy.settings/Settings/Gui/DefaultOverview` setting, as configured by gui-v1, to avoid adding a new setting. In this case, `OverviewHub` = Brief page, and `OverviewTiles` = Overview page. If the existing setting is used, then only use these two values, otherwise gui-v1 cannot use the setting.

chriadam, 2024-01-31:
Not for 1.0.0, marking as low_prio.

kwindrem, 2024-01-31:
There are times when I'll leave something other than the overview screen active. For example, leaving a temps page up so I can quickly check temps during the night. So I'm opposed to automatically changing what's displayed.

gui-v1 puts the name of the qml object to go to in the default overview parameter so it can be anything you want it to be. Current behavior is to select the first overview page if the value in the parameter can't be found so this should be fine if gui-v2 puts some other string in that parameter.

mr-manuel, 2024-05-01:
Is it possible to use this as startup screen after a reboot or after the wasm build is loaded? Right now every time the brief page is shown.

blammit, 2024-09-23:
Have discussed an alternative solution that will obsolete this one: https://github.com/victronenergy/gui-v2/issues/1496

Closing.

closed 2024-09-23 by blammit