victronenergy/gui-v2 #723: Align empty state handling between views (where it makes sense)

jpetrell, 2024-02-08:
Before setup the gui-v2 is initially (but quite understandably) quite bare, e.g. Control cards and Leves pages are totally empty. Some views have empty state placeholders (e.g Notifications page's green check icon + "_No current alerts_" or ESS settings page's "_No ESS Assistant found_"), but most don't. Fully empty page is not necessarily wrong even if also not that helpful either, could consider adding some kind of friendly actions to the most likely next steps to help the user populate the view (or even just friendly placeholder graphic+label describing what to expect here once the setup is complete).
In some cases the setting could be hidden

Some empty states horizontally center the placeholder label (Tank pump setting), some left align (Notifications, ESS settings). Generator start/stop setting uses list item background, others just display a plain label. Go through the views, implement the empty state handling consistently.

jpetrell, 2024-04-16:
Different empty state layouts:
[image]

The simple left-aligned label placeholder (see Wi-Fi example above) is used elsewhere too:
- Tank pump settings "Tank pump start/stop function is not enabled. Go to relay settings and set function to \"Tank pump\"."
- Inverter settings "No alarms to be configured"
- Tank shape settings "No custom shape defined. You may define one with up to ten points. Note that 0% and 100% are implied."
- Battery fuse (Lynx?) settings "No information available, see previous page for Distributor status."
- Certain device settings pages in "multi" or "charger" types: "The settings details for this device will be provided in a future update."
- Relay temperature sensor settings "No relay is configured to be activated by temperature. Go to the relay settings page located in the main settings menu and set the relay function to \"Temperature\"."
- ESS settings "No ESS Assistant found"
- Energy meter settings "No energy meters found. Note that this menu only shows Carlo Gavazzi meters connected over RS485. For any other meter, including Carlo Gavazzi meters connected over ethernet, see the Modbus TCP/UDP devices menu instead."

blammit, 2025-03-17:
Note, Serj provided some empty state views in Figma a few months back, so this can be investigated further.