victronenergy/gui-v2 #728: Opening Wi-Fi access point settings randomly crashes on gui-v2

jpetrell, 2024-02-08:
Go to Settings -> Wi-Fi -> Open access point. Now go back and open it again. After 3-12 tries the gui-v2 crashes and the terminal shows error

```
free(): invalid pointer
Aborted

```

Tested on Ekrano GX, firware v.3.20~46, application version v0.2.9.

DanielMcInnes, 2024-02-26:
@jpetrell , do you mean: Settings -> WiFi -> toggle the 'Create access point' switch?

jpetrell, 2024-03-05:
> do you mean: Settings -> WiFi -> toggle the 'Create access point' switch?

No sorry. I meant repeatedly opening the Wi-Fi networks page. Updated the description now to be more accurate.

jpetrell, 2024-03-05:
Ran venus-gui-v2 with gdbserver. The stack trace didn't give much, other than the gui process throws SIGABRT due to critical error (assert, unhandled throw).

```
Thread 1 "venus-gui-v2" received signal SIGABRT, Aborted.
0x441417e8 in raise () from target:/lib/libc.so.6
(gdb) bt
#0  0x441417e8 in raise () from target:/lib/libc.so.6
#1  0x4412bc84 in abort () from target:/lib/libc.so.6
#2  0x441796bc in ?? () from target:/lib/libc.so.6
```

The only asserts on Victron side are inside VeQItem, but they are not called. Though "free(): invalid pointer" seems to suggest something somewhere is trying to delete an object that has already been freed from the memory. Will keep looking.

Bumbed into another page stack issue #832 while trying to reproduce this one.

jhofstee, 2024-03-14:
For what it is worth, this (likely) happened to me as well:

```
[  979.930326] potentially unexpected fatal signal 6.
[  979.935201] CPU: 3 PID: 1439 Comm: venus-gui-v2 Tainted: G           O      5.10.109-venus-15 #1
[  979.944005] Hardware name: Allwinner sun8i Family
[  979.948735] PC is at 0x454217e8
[  979.951925] LR is at 0xbec6e698
[  979.955069] pc : [<454217e8>]    lr : [<bec6e698>]    psr: 00070010
[  979.961346] sp : bec6e698  ip : 00000000  fp : bec6e920
[  979.966600] r10: 45515ef0  r9 : 4552e2bc  r8 : 00000001
[  979.971824] r7 : 000000af  r6 : bec6e8d8  r5 : bec6e8d8  r4 : 00000006
[  979.978374] r3 : 00000008  r2 : 00000000  r1 : bec6e698  r0 : 00000000
[  979.984914] Flags: nzcv  IRQs on  FIQs on  Mode USER_32  ISA ARM  Segment user
[  979.992155] Control: 10c5387d  Table: 41dcc06a  DAC: 00000055
[  979.997902] CPU: 3 PID: 1439 Comm: venus-gui-v2 Tainted: G           O      5.10.109-venus-15 #1
[  980.006677] Hardware name: Allwinner sun8i Family
[  980.011404] [<c010d19c>] (unwind_backtrace) from [<c010a2c8>] (show_stack+0x10/0x14)
[  980.019150] [<c010a2c8>] (show_stack) from [<c0658cc0>] (dump_stack+0x98/0xac)
[  980.026375] [<c0658cc0>] (dump_stack) from [<c01342a0>] (get_signal+0x794/0x79c)
[  980.033774] [<c01342a0>] (get_signal) from [<c0109984>] (do_work_pending+0x100/0x550)
[  980.041602] [<c0109984>] (do_work_pending) from [<c01000cc>] (slow_work_pending+0xc/0x20)
[  980.049769] Exception stack(0xc1bdbfb0 to 0xc1bdbff8)
[  980.054820] bfa0:                                     00000000 bec6e698 00000000 00000008
[  980.062992] bfc0: 00000006 bec6e8d8 bec6e8d8 000000af 00000001 4552e2bc 45515ef0 bec6e920
[  980.071162] bfe0: 00000000 bec6e698 bec6e698 454217e8 00070010 00000002
```

closed 2024-03-20 by jpetrell

jpetrell, 2024-03-20:
Opened https://github.com/victronenergy/gui/pull/22 for gui v1.

commit e9a252e4 (2024-03-19): Avoid crash when QML tries to delete global CmAgent

Fixes #728.
commit 5860e3e3 (2024-03-20): Avoid crash when QML tries to delete global CmAgent

Fixes #728.