victronenergy/gui-v2 #734: On iPhone, settings menu and more is broken per v0.2.9

mpvader, 2024-02-09:
Discussed here: https://victrondevelopment.slack.com/archives/C020QG3JVEW/p1707458487757799

Matthijs can reproduce.

Agreed first step is to test a build that was made with Qt 6.6 instead of Qt 6.5.something. Afaik Dirk-Jan already has that process working

dirkjanfaber, 2024-02-09:
@mpvader : Can you reproduce it on [this site](https://acceptancevrm.victronenergy.com/installation/237451/dashboard)?

That one is running the gui with a Qt 6.6.1 build. (using  [this](https://github.com/victronenergy/gui-v2/tree/faberd/qt-6.6.1-emscripten-3.1.37) branch and the files from [this action](https://github.com/victronenergy/gui-v2/actions/runs/7843682943)).

mpvader, 2024-02-09:
Hi @dirkjanfaber that solves it indeed.

So then plan remains the same: await Lucas from El Nino to finish and deploy his preps for Qt 6.6, and then switch the webassembly build to that; right?

dirkjanfaber, 2024-02-10:
I believe that it is only a matter of deploying for Lucas. I've just put this build on CDN and confirm that running that also works on acceptance VRM via the remote console 2 link. Meaning that both the Qt 6.6.1 as the Qt 6.5.2 builds work on all tested cases:

| Build  | local | CDN |
| --- | --- | --- |
| Qt 6.5.2 | ✅ | ✅  |
| Qt 6.6.1 | ✅ | ✅ |

With local I mean that VRM downloads the .wasm file from the local device. With CDN I mean that VRM downloads the .wasm file from the CDN.

So if Chris and Daniel agree (see https://github.com/victronenergy/gui-v2/issues/677), my branch can be merged and the code from acceptance VRM can be deployed on regular VRM. Both old and new builds work.

mpvader, 2024-02-10:
Nice

blammit, 2024-02-19:
PRs merged, closing.

closed 2024-02-19 by blammit