victronenergy/gui-v2 #737: Device List: com.victronenergy.genset.xxx Engine parameter inconsistencies

philipptrenz, 2024-02-09:
I found some inconsistencies regarding generator engine parameters:
- Load should be displayed rounded without a decimal place
- Load unit "%" is missing
- Oil pressure unit "kPA" is missing
- Coolant temperature unit should be "°C" instead of "°"

[image]
[image]

philipptrenz, 2024-02-09:
Furthermore, I want to propose two changes that go beyond gui-v1:
- Adding the `/Engine/OilTemperature` dbus path element, unit "°C", rounded to no decimal place
- Changing "Operating time" formatting to always display in hours

I have been asked the latter several times by customers; a unit other than hours is not typical for generators.

blammit, 2024-02-19:
@jepefe could you comment on the `/Engine/OilTemperature` and "Operating time" suggestions please?

Also, in gui-v2 the Overview page and Levels pages just show the degrees symbol, e.g. "15°" instead of "15°C", and this formatting extends to the settings pages as well. Is this okay, or should it show the longer "15°C" form in the settings pages, to be consistent with gui-v1?

jepefe, 2024-02-19:
@blammit 
The full unit must be shown so "15°C" everywhere and for this page the decimal can be removed.
Operating time can be displayed in hours instead of formatted time string. This is probably more convenient for operators.

blammit, 2024-02-19:
@jepefe thanks, and what about adding `/Engine/OilTemperature` as Philipp suggested?

philipptrenz, 2024-02-20:
I can add it to gui-v1, although I think it is not worth a back port and thus will be released not before v3.3 – when v1 has maybe already been replaced by v2.

jepefe, 2024-02-20:
@blammit 
It can be added, make it show conditionally when the reading is available.

blammit, 2024-03-05:
> Furthermore, I want to propose two changes that go beyond gui-v1:
>
> * Adding the `/Engine/OilTemperature` dbus path element, unit "°C", rounded to no decimal place
> * Changing "Operating time" [...]

@philipptrenz if displaying the operating time in hours, should it be displayed as "hh:mm" if there are minutes? Or what would you suggest?

philipptrenz, 2024-03-05:
@blammit I also asked myself the same question before. I guess a higher resolution does not hurt, so I would opt for `XXXh YYm`

blammit, 2024-03-07:
Fixed by https://github.com/victronenergy/gui-v2/pull/830

closed 2024-03-12 by blammit