victronenergy/gui-v2 #738: Add support for Orion XS

mpvader, 2024-02-10:
@ThiemoVanEngelen , can you make a full list for the gui-v2 dev team of what needs doing? Both when on D-Bus as `.dcdc` as well as `.alternator`. 

Things that come to mind here are:

- Device-list entry as alternator will need some updating
- Device-list entry as dcdc converter will need some updating
- Page that shows when clicking on alternator box on overview will need updating (but that might be the same code as used for the device list)
-  Calculation of DC System box might need a change, for that I'll make a separate issue.

ThiemoVanEngelen, 2024-02-12:
- [ ] Create a new page as Device-list entry for .alternator service with product id 0xA3F0..0xA3FF. (PageAlternatorModel.qml from gui-v1 can be used as example). The following items are used for the OrionXS:
  - Switch (`/Mode`)
  - Input Voltage / Current / Power (`/Dc/In/V`, `/Dc/In/I`, `/Dc/In/P`)
  - Output Voltage / Current / Power (`/Dc/0/V`, `/Dc/0/I`, `/Dc/0/P`)
  - State (`/State`)
  - Network status (`/Link/NetworkStatus`)
  - Error (`/ErrorCode`)
  - Overall history (`/History/Cumulative/User/....`)
  - Charge cycle history (`/History/Cycle/.....`)
  - Device (The regular device menu)
- [ ] Create a new page as Device-list entry for .dcdc services. (PageDcDcConverter.qml from gui-v1 can be used as example). The following items are used for the OrionXS:
  - Switch (`/Mode`)
  - Input Voltage / Current / Power (`/Dc/In/V`, `/Dc/In/I`, `/Dc/In/P`)
  - Output Voltage / Current / Power (`/Dc/0/V`, `/Dc/0/I`, `/Dc/0/P`)
  - State (`/State`)
  - Error (`/ErrorCode`)
  - Device (The regular device menu)
- [ ]  Add the same logic as the Solar Yield box when there are multiple .alternator services. So show a device list with the devices and the main information: output voltage, current and power

ThiemoVanEngelen, 2024-02-12:
Do we also want people to be able to press the DC loads box and see al know DC loads?

mpvader, 2024-02-12:
@ThiemoVanEngelen can you make sure there is a way for gui-v2 team to test this? (make sure there is a recording, and where to find that).

And @jpetrell in case you're going to work on this, make 100% sure to look at the relevant equivalent pages in the gui-v1 repo. Don't overlook that comment from Thiemo.

ThiemoVanEngelen, 2024-02-14:
Here is dbus recording of an OrionXS as com.victronenergy.alternator.ttyS6:
[OrionXS.zip](https://github.com/victronenergy/gui-v2/files/14277272/OrionXS.zip)

It starts not charging but after a short while will start charging for a while and then stop again

ThiemoVanEngelen, 2024-02-14:
Here is a dbus recording of an Orion XS as com.victronenergy.dcdc.ttyS6:
[OrionXS_dcdc.zip](https://github.com/victronenergy/gui-v2/files/14277918/OrionXS_dcdc.zip)

This also starts in Off, goes to PSU mode after a while and after a long time goes back to Off.

blammit, 2024-02-19:
@ThiemoVanEngelen are there further details to be provided, or is this ready to implement?

ThiemoVanEngelen, 2024-02-26:
@blammit I think all the information is here.

blammit, 2024-02-27:
> Do we also want people to be able to press the DC loads box and see al known DC loads?

@ThiemoVanEngelen can you confirm the answer to this? And if yes, then I assume this means `.dcdc` services should appear in the list when you click on the DC Loads box? (Currently it only shows `dcload` and `dcsystem` services.)

mpvader, 2024-02-28:
> @ThiemoVanEngelen can you confirm the answer to this? And if yes, then I assume this means .dcdc services should appear in the list when you click on the DC Loads box? (Currently it only shows dcload and dcsystem services.)

@blammit yes pls do that. thanks.

blammit, 2024-02-28:
> @ThiemoVanEngelen can you confirm the answer to this? And if yes, then I assume this means `.dcdc` services should appear in the list when you click on the DC Loads box? (Currently it only shows [...]

Discussed with Matthijs, the answer is yes, and we should show `dcdc` services in this list.

blammit, 2024-03-08:
@ThiemoVanEngelen Supporting multiple alternators in the Overview alternator box complicates things a bit. Some questions:

- When displaying the "total" for the alternator box: if showing watts, I assume this can be the sum total of the `/Dc/0/Power` for all alternators. If showing amps, what does this show? The `/Dc/0/Current` of the first found alternator?
- Regarding the electron flow line that connects the alternator box to the battery: currently the animation flows towards the battery if the alternator `/Dc/0/Power` is > 30. If there are multiple alternators, should the the sum total of the `/Dc/0/Power` for all alternators be used to calculate this instead?

blammit, 2024-03-13:
If there are multiple alternators, can sum the total output current for the display.

closed 2024-03-20 by blammit