victronenergy/gui-v2 #739: Define and document logic for DC Loads box

mpvader, 2024-02-10:
- [ ] Define a good solution, I already [added a comment](https://docs.google.com/document/d/1yzHSpZoeZ0WOJJPhjivbhhD_iz37PaZtnZ9GpqVPh2M/edit?disco=AAABGWWhZiY) to our How to design a UI document.
- [ ] Modify the implementation

kwindrem, 2024-02-10:
While a DC shunt feeding the loads is the "right" way to design a system, do not forget about the previous way DC System was done. Essentially, it's what ever is left over after all the "known" sources and loads are accounted for.

The document referenced above does not mention this but the "calculated" value is important for systems that do not have a DC Loads shunt.

Work on the calculation in systemcalc is also needed to incorporate some of the sources not previously included in the calculation. gui-v2 also shows more DC sources than gui-v1 so the calculations may differ. In fact, gui-v2 may not even be using the system calc's DS System value and that's probably correct.

mpvader, 2024-02-10:
Hey @kwindrem , yes sure, thanks. The solution has to work for both methods, calculated loads and systems where a shunt is installed that measures the loads.

ReinvdZee, 2024-03-27:
I've added a small description for the DC loads box in the [document](https://docs.google.com/document/d/1yzHSpZoeZ0WOJJPhjivbhhD_iz37PaZtnZ9GpqVPh2M/edit?disco=AAABGWWhZiY) 

### Systemcalc
[Systemcalc](https://github.com/victronenergy/dbus-systemcalc-py/blob/master/dbus_systemcalc.py#L746-L794) calculates/estimates `/Dc/System/Power` with the following procedure:

1. If services of type `com.victronenergy.dcsystem` available, then `/Dc/System/Power` is equal to the sum of the `/Dc/0/Power` field of all of these services.
2. Else, if there is a battery service (shunt, BMV, etc), then `/Dc/System/Power` is equal to the sum of the power of all sources - battery power
3. Else if there is `solarchargers_loadoutput_power` (not sure what this is), then `/Dc/System/Power` is equal to `solarchargers_loadoutput_power`

@Izak, for case 2, there is a comment in systemcalc that states that for gui-v2, `alternator_power` (`/Dc/Alternator/Power`) must be subtracted as well. However, wouldn't we need to add this one instead of subtracting it? Because `/Dc/System/Power` should be equal to the sum of all sources (including the alternator) minus the power flowing into the battery, right?

izak, 2024-03-27:
@ReinvdZee It has been a long time since I wrote that. From a cursory inspection you are probably right, it needs to be added.

Do note the sign of some of the variables in that expression. `vebus_power` and `inverter_power` are negative values when power is being consumed which is why they are added and not subtracted.

Ideally also add/modify a unit test for it, that helps to know that it is right and that it stays right.

ReinvdZee, 2024-04-04:
In GUI-V1, `/Dc/System/Power` is being used by the ["DC SYSTEM" tile ](https://github.com/victronenergy/gui/blob/master/qml/OverviewTiles.qml#L231C5-L248C6). In order to not change the behavior of GUI-V1, we should not modify `/Dc/System/Power` for the sake of GUI-V2. 

Since the box is called "DC loads" in GUI-V2, I suggest creating two new paths for GUI-V2 to use:
- `/Dc/Loads/Power`
- `/Dc/Loads/Current` 

These paths would hold the power and current of the DC loads, taking the alternator into account. @izak, @jepefe, what do you think?

And concerning this:
> Ideally also add/modify a unit test for it, that helps to know that it is right and that it stays right. python-nosetest3 is used for the unit tests, you can apt-get install that.

I will modify the unit tests to test the calculation of the DC loads box values.

kwindrem, 2024-04-04:
The DC System power calculations in systemcalc aren't really correct even for gui-v1. The calculations are supposed to remove all "known" DC sources and loads. What is left is reported as DC System power. However, only SOME of those sources and loads are removed while others are not removed. gui-v1 has tiles for only a small number of DC devices: battery, PV charger, inverter/charger. There are notes in systemcalc saying FIXME.

I understand that gui-v2 attempts to show all or most of the DC devices in separate tiles. In any case, what's "left over" for DC System will depend on the other DC devices actually shown on the GUI !!! This suggests the GUI itself needs to calculate the value for the DC System tile.

The shift in name from DC System to DC Loads is most likely due to the typical case that the "what's left" is DC loads in an RV or boat. However, I'm not sure that assumption holds in all cases. GuiMods allows renaming the tile so the system designer can provide a meaningful name.

Systemcalc also provides a "measured" value for DC System power and current if provided by a shunt programed for DS System (but not DC Loads). It should probably include both in its measured accumulation.

The "what's left" calculation for DC system power is iffy at best. The best thing to do would probably allow the system designer to specify which parts of the system to include in the DC System power calculations. This would be a benefit for gui-v1 as well. The DC sources and loads that need to be included/excluded are:
AC Charger, DC-DC charger, Alternator, Wind generator, Fuel Cell, Motor drive.
Battery, inverter/charger and PV charger are always present in the GUI as far as I can tell.

ReinvdZee, 2024-04-12:
Discussed with @izak today, these paths will be added to systemcalc for gui-v2 to use:

```
/Dc/Loads/Power
/Dc/Loads/Current
```

We will not touch the path used by gui-v1

mpvader, 2024-04-18:
I asked Izak to plan this soon

ReinvdZee, 2024-04-23:
Last week I discussed this with @izak, and we agreed on adding those two paths:

>/Dc/Loads/Power
/Dc/Loads/Current

So it only needs to be implemented in systemcalc.

izak, 2024-05-17:
After more discussion, we decided not to add more dbus paths.

/Dc/System/Power has always been an estimate for the DC loads (or that measured with an appropriately configured shunt acting as a dcsystem). It is technically broken on GUIv1 because it shows alternators in the "dc power" box, but does not consistently do the same for other DC sources, nor does it properly account for systems where the dc system is measured by a shunt (and the alternator would be on the supply side of the shunt).

The correct answer is therefore to revert to the planned implementation, which was reverted for the 2.90 release as an interim measure.

In addition we will then add /Dc/System/Current as well.

closed 2024-05-17 by jhofstee

blammit, 2024-05-20:
> In addition we will then add /Dc/System/Current as well.

@izak If that is added, then sounds like gui-v2 should be updated to show this value when user prefers to see amps? Currently if amps is preferred, it shows the value of `/Dc/System/Power /  /Dc/System/Voltage`.

izak, 2024-05-20:
@blammit In systemcalc I do exactly the same thing anyway, I divide the power by the voltage. But it probably is cleaner to use this value instead of calculating it yourself.