victronenergy/gui-v2 #743: Add support for Thai and Chinese for webassembly builds

dirkjanfaber, 2024-02-12:
The current webassembly builds lack support for Thai and Chinese because of font rendering problems. 

The needed fonts are `NotoSansThai.ttf` and `DroidSansFallback.ttf`. Both do reside in `/usr/lib/fonts` on a GX device, but might need to be made accessible for web retrieval (can be done with creating a symbolic link from place on the webserver).

See https://github.com/victronenergy/gui-v2/pull/673#issuecomment-1931533857 for further info.

mpvader, 2024-09-26:
hey @blammit , and @chriadam and @DanielMcInnes : this issue was found by our Chinese distributor, he reported it yesterday.

And to be honest I had forgotten about this issue a bit. Its not a blocker for official release of Venus OS with gui-v2, but for sure its a nice to have soon after.

Or at a minimum, we can change the text of this popup to say that Thai and Chinese are not supported yet when accessing this UI, and therefore its available in English only; and expected 2025 Q1. Or something like that.

I have no idea how much work there is in this one; and if it needs discussion with people. For example: if we'd statically host those files on a url, same domain for example as where gui-v2 is downloaded from, would that work and be sufficient for you?

Then we'd arrange hosting on both VRM and locally on the GX and problem easily solved?

blammit, 2024-09-27:
> I have no idea how much work there is in this one; and if it needs discussion with people. For example: if we'd statically host those files on a url, same domain for example as where gui-v2 is downloaded from, would that work and be sufficient for you?

> Then we'd arrange hosting on both VRM and locally on the GX and problem easily solved?

Yes if it's on the same domain then that should be sufficient. For local wasm the font file can be put in `/var/www/venus/gui-beta` and then for VRM access it would have to be some pre-defined location that can be loaded by gui-v2.

mpvader, 2024-09-27:
thanks, then thats what we'll do.

@jhofstee can you make those two fonts available on the webserver? for people accessing http://[IP]/gui-v2. 

I'll arrange the fonts to be on VRM.

jhofstee, 2024-09-27:
I would prefer to have them in http://[ip/fonts instead, so it can be shared. Or does that cause any issue?

chriadam, 2024-10-02:
Any path is fine, so long as it's served by the nginx webserver we can load it

jhofstee, 2024-10-02:
@blammit there is nothing blocking this issue, it is trivial to copy / symlink fonts.

jhofstee, 2024-10-02:
So /fonts it is, with these fonts at the moment:
```
DejaVuSans-Bold.ttf
DejaVuSans.ttf
DejaVuSansMono.ttf
DroidSansFallback.ttf
NotoSansThai.ttf
```

blammit, 2024-10-02:
@jhofstee great, thanks!

mpvader, 2024-10-03:
a short update:

- font work by Jeroen is included per Venus OS v3.50~28; not available yet; its pending a CI issue.
- I’ve asked the VRM team to do the same, here: https://victrondevelopment.slack.com/archives/C06GFC1B03U/p1727929403160159

dirkjanfaber, 2024-10-07:
The fonts are now available via VRM as well under `https://updates.victronenergy.com/fonts/`.  Mimicing the location on the Venus device, so you should be able to find the font under `/fonts/${font}.ttf`, no matter if the wasm is running on Venus or on VRM. 
Also added a symbolic link on the server so that `../fonts/${font}.ttf` works as well. 

At the moment serving the files: `DejaVuSans-Bold.ttf  DejaVuSans.ttf  DejaVuSansMono.ttf  DroidSansFallback.ttf  NotoSansThai.ttf`

chriadam, 2024-10-16:
With the above PR, gui-v2 now loads appropriate font files from the webserver for Chinese and Thai. I noticed while testing the above PR that the fallback English catalogue is missing some translations.  I've asked Daniel to look into that, since he solved a similar issue recently with the Chinese catalogue.

chriadam, 2024-10-22:
Fixed.

closed 2024-10-22 by chriadam

mpvader, 2024-10-22:
Nice!