victronenergy/gui-v2 #746: Support tanks without "capacity" and "remaining" values.

blammit, 2024-02-13:
Some tanks may not have values for both the "capacity" and "remaining" values. In this case, the "level" value should be used to calculate the displayed level in the Tanks tab, instead of trying to calculate it from the capacity/remaining.

DanielMcInnes, 2024-02-26:
martin
  12 days ago
@Bea Lam
 Is it a known issue that the % is not properly calculated for tanks that are grouped together?
Has been reported by a tank sensor integration partner. Their tank senders don't have a capacity set and thus no absolute level.
[image]

martin
  12 days ago
On my system it looks like this in gui v1 and 2 (only one sensor properly connected)
[image]

martin
  12 days ago
Remaining & Capacity topics are null
[image]

Bea Lam
  12 days ago
I didn’t realise that both remaining and capacity could be missing. Thanks, I’ve created https://github.com/victronenergy/gui-v2/issues/746 to fix this.
I guess in this case, the bottom label can be hidden, instead of showing some nonsense like 0.000/0.000m3.
Note that you might see some tank volume and unit conversion issues on older releases too, there were a bunch of tank fixes in the recent v3.30~3 release.
[image]

DanielMcInnes, 2024-02-27:
Adding 'blocked' tag - I need the site with the issue to come online to test my changes.

blammit, 2024-03-13:
Removing blocked tag, we could mimic the conditions locally for now.

mpvader, 2024-03-13:
@martinbosma what brand or type tank senders is this about? can you give me some background to understand this one better?

martinbosma, 2024-03-13:
Garnet, connected via RV-C.

kwindrem, 2024-03-13:
In addition to SeeLevel RV-C not having capacity, both SeeLevel RV-C and N2K do not support custom names. This makes it difficult to differentiate tanks as displayed on the GX device. While SeeLevel RV-C and N2K don't support two identical tank types, Soul does. I would expect Soul to not report capacity or names as well because it uses RV-C.

I see no reason Venus OS can't add local /Capacity, /CustomName parameters and calculate /Remaining locally if the sensor system doesn't report them. If values are received from the sensor system they would simply overwrite any values set locally.

A discussion about this some time ago suggested that the "right" way is for the sensor to support these parameters so that all control systems show these senors receive the same information. That may be true but results in missing information which could easily be set on each control system once during system configuration.

blammit, 2024-03-13:
Note that gui-v2 will calculate the equivalent /Capacity if the /Level and /Remaining are available, though perhaps this could be done at a lower level instead for convenience, to make it available elsewhere.

commit 1272f88a (2024-03-15): Allow for missing /Level tank values in grouped tank gauges

When showing the average level for a grouped tank gauge, and each
tank has a /Level value, use the /Level values to calculate the
average level. This is preferable to calculating it from the total
/Remaining and /Capacity values as those may not be available.

Fixes #746
commit d926c711 (2024-03-15): Allow for missing /Level tank values in grouped tank gauges

When showing the average level for a grouped tank gauge, and each
tank has a /Level value, use the /Level values to calculate the
average level. This is preferable to calculating it from the total
/Remaining and /Capacity values as those may not be available.

Fixes #746
closed 2024-03-15 by jpetrell