victronenergy/gui-v2 #754: Date/time setting issues

ReinvdZee, 2024-02-14:
There is currently no option to set the date manually:

[image]

Also, when setting the time to some value, the hour is off. E.g., when setting the time to 10:30, the result is 9:30. This problem does not occur when the GX device timezone is set to the same timezone as the browser (tested with GMT+1).

chriadam, 2024-03-19:
I'll look into this.  The base problem is that WebAssembly builds don't support timezones other than UTC and the current local timezone, due to the fact that it's the only timezones available in the browser context.  See https://bugreports.qt.io/browse/QTBUG-91441 and previous investigation in https://github.com/victronenergy/gui-v2/pull/307 for more information.

But if we're simply adjusting values via a spinbox control, then there need not be any conversion to a date object at all, thus no need to modify timezone etc.

chriadam, 2024-03-19:
The `N/<portal_id>/platform/0/Device/Time` topic contains a number which appears to be the number of seconds since epoch, but in the device's local timezone.  Is there also some topic where the device publishes its current local timezone offset?

The problem is that in gui-v2 webassembly builds, we don't have real access to timezone information.  The timezone selections in gui-v2 consist of strings only, e.g.: TzAustraliaData looks like

```
import QtQuick

ListModel {
	id: tzCity
	ListElement { display: QT_TR_NOOP("Tasmania Standard Time"); city: "Hobart"; caption: "(GMT +10:00) Hobart" }
	ListElement { display: QT_TR_NOOP("E. Australia Standard Time"); city: "Brisbane"; caption: "(GMT +10:00) Brisbane" }
	ListElement { display: QT_TR_NOOP("AUS Eastern Standard Time"); city: "Sydney"; caption: "(GMT +10:00) Canberra, Melbourne, Sydney" }
	ListElement { display: QT_TR_NOOP("Cen. Australia Standard Time"); city: "Adelaide"; caption: "(GMT +09:30) Adelaide" }
	ListElement { display: QT_TR_NOOP("AUS Central Standard Time"); city: "Darwin"; caption: "(GMT +09:30) Darwin" }
	ListElement { display: QT_TR_NOOP("W. Australia Standard Time"); city: "Perth"; caption: "(GMT +08:00) Perth" }
}
```

These strings are not consumable programmatically, really.  I don't want to manually try to add offset information for each timezone (since it will invariably require DST transitions etc etc so super difficult to do properly without using olsendb data etc).

Instead, if the device published a topic like: `N/<portal_id>/platform/0/Device/TimeZoneOffset` with current number of seconds offset from UTC, then in gui-v2 webassembly we could do:

1) read /Device/Time value
2) read /Device/TimeZoneOffset value
3) convert /Device/Time value to UTC by adding /Device/TimeZoneOffset value
4) allow user to select a new (browser local) time via the spinners
5) convert that browser local time to UTC
6) convert that UTC time to device-local time by subtracting the /Device/TimeZoneOffset value
7) set that value as the newly selected /Device/Time value.

mpvader, 2024-03-28:
Hi Chris, we discussed this in the meeting today. And we or I don't understand what the timezone of the browser has to with it.

The path, ` N/<portal_id>/platform/0/Device/Time`, now has seconds since epoch in UTC. Shall we change that to seconds since epoch in device-selected time zone?

Then gui-v2 has nothing to do on time zone conversions wrt this edit box.

please confirm, and then we can change venus-platform.

chriadam, 2024-04-02:
Seconds since epoch in UTC is fine :-)  No other changes are necessary in that case, except in gui-v2.  Thanks.

chriadam, 2024-04-18:
Fix has been merged.

closed 2024-04-18 by chriadam