victronenergy/gui-v2 #765: RS products: Update alarms

ReinvdZee, 2024-02-16:
Add support for some recently added alarms to RS products.
See issue: victronenergy/venus-private#288

**Solar charger**
The "High temperature" and "Short circuit" alarms should be added to the "Solar charger" page (afaict pages/solar/SolarChargerAlarmsAndErrorsPage.qml). 

Related commit on gui-v1:
https://github.com/victronenergy/gui/commit/49512faa53246f2a7c23192c8beaea26a1b7cccd

**Inverter RS / Multi RS**
afaict, at least the "Short-circuit" alarm is missing here:
https://github.com/victronenergy/gui-v2/blob/main/src/gui-v1/alarmbusitem.cpp#L97-L111
I could also not find the "lowsoc" alarm for the Multi RS.

Related commit to gui-v1
https://github.com/victronenergy/gui/commit/81fff9c6efc4b31c6713d6cc12e49b740a6b9a1b

blammit, 2024-05-13:
> Inverter RS / Multi RS
> afaict, at least the "Short-circuit" alarm is missing here:
> https://github.com/victronenergy/gui-v2/blob/main/src/gui-v1/alarmbusitem.cpp#L97-L111
> I could also not find the "lowsoc" alarm for the Multi RS.

Reopening issue as this part has not been resolved yet.

jpetrell, 2024-06-06:
> afaict, at least the "Short-circuit" alarm is missing here:

Anything else still needed here? Short-circuit alarms were added in separate task. 

> I could also not find the "lowsoc" alarm for the Multi RS.

Maybe this one? ^

blammit, 2024-06-08:
> I could also not find the "lowsoc" alarm for the Multi RS.

I'll check, I added the Multi RS recently so that might be up-to-date already.

blammit, 2024-06-09:
> I'll check, I added the Multi RS recently so that might be up-to-date already.

Verified this was done as part of https://github.com/victronenergy/gui-v2/pull/1183.

commit 175df23c (2024-04-30): Add high temperature and short circuit Solar Charger alarms

Fixes #765.
commit aade44a1 (2024-05-02): Add high temperature and short circuit Solar Charger alarms

Fixes #765.
closed 2024-06-09 by blammit