victronenergy/gui-v2 #777: When set to Amps, certain readings keep increasing

mpvader, 2024-02-19:
Reported here: https://community.victronenergy.com/questions/262520/venus-os-v3304-available-for-testing.html

[image]

[image]

jpetrell, 2024-02-28:
The Brief page gets the amps from a component called `SystemAc`, which calls the amps "total current" of an one-phase system. Whenever either `"/Ac/ConsumptionOnInput/L[n]/Current"` or `"/Ac/ConsumptionOnOutput/L[n]/Current"` values update the `SystemAc`'s current property gets updated by summing up the existing current with the updated currents.

It is easy to disable the accumulation, but I don't really understand what exactly is the correct behavior:

```
data/SystemAc.qml
-    _firstPhaseCurrent = Units.sumRealNumbers(_firstPhaseCurrent, data.current)
+    _firstPhaseCurrent = data.current
```

Will sync tomorrow with Rein.

kwindrem, 2024-02-28:
I recall a recent discussion on this subject and agree that currents should no be summed across multiple phases. The only valid information is individual leg currents so these need to be shown separately as a numeric value and where gauges are used, as multiple bars on the gauge.

Summing currents implies that somehow the maximum allowed current is the sum of the maximums of all the phases when in fact there is no such limit. A potential overload can potentially be hidden by looking at a summed current if one leg's current is substantially larger than the others.

ReinvdZee, 2024-02-29:
@jpetrell, 

The top right gauge displays the value of `Ac/ConsumptionOnOutput/L1/Current`. This is also mentioned [here](https://github.com/victronenergy/venus/issues/1217#issuecomment-1961670898)

@jepefe, @izak, can you confirm?

PR #817 (2024-03-07): Don't accumulate current shown on Brief&Overview pages
You can change to amps from Display & Language -> Units -> Electrical power display -> Select Current (Amps).

Amperes are only shown on an one-phase system. Changed to only display the output current (Jesus commented last week that the top right gauge should follow `Ac/ConsumptionOnOutput/L1/Current` in  https://github.com/victronenergy/venus/issues/1217).

Fixes #777.
commit 75e64492 (2024-03-07): Don't accumulate current shown on Brief&Overview pages

Fixes #777.
closed 2024-03-07 by jpetrell