victronenergy/gui-v2 #786: Add support for additional alternators

ReinvdZee, 2024-02-21:
Add support for the following alternator types:

1. MG AFC: 0xB0F0 --> https://github.com/victronenergy/gui/commit/e7bc072c358da7a6813d98a0d3fcc2d6b7744380
2. Generic alternator: 0xB091 --> https://github.com/victronenergy/gui/commit/700213aa1014618832e37a40b46ffe56f626150c

Furthermore, invalid items should be hidden (if this is not already the case): https://github.com/victronenergy/gui/commit/d5c2b3c010dc0a8bbe458db5517cc779bb3647c4
Which also may require this change: https://github.com/victronenergy/gui/commit/afa657536480a28459a8fb1f74bf976cd02540c8 or something similar.

nmbath, 2024-03-13:
Could we also include the product codes for Integral (0xB092) https://github.com/victronenergy/gui/commit/aafd7a24752b2c420142e427c9accab430e96e3f

jpetrell, 2024-04-10:
> MG AFC: 0xB0F0 --> https://github.com/victronenergy/gui/commit/e7bc072c358da7a6813d98a0d3fcc2d6b7744380
> Generic alternator: 0xB091 --> https://github.com/victronenergy/gui/commit/700213aa1014618832e37a40b46ffe56f626150c
> Integral (0xB092) https://github.com/victronenergy/gui/commit/aafd7a24752b2c420142e427c9accab430e96e3f

These have already been added elsewhere.

jpetrell, 2024-04-12:
Done.

closed 2024-04-12 by jpetrell