victronenergy/gui-v2 #809: The screen is impossible to read on the lowest display brightness setting

jpetrell, 2024-02-27:
Go to Settings -> Display & Language -> Disable Adaptive brightness -> Move Brightness slider to the lowest value.

Expected result: The display becomes dim, but still stays more or less legible (unless you are trying to use it e.g. direct sun light).

Actual result: The display becomes so dark that it is effectively blanked. If you accidentally leave the Display & Language page you might not be able to find your way back and put the display back on using the GUI.

Tested on Ekrano GX. Doesn't reproduce on gui v1.

blammit, 2024-02-28:
The min value should be fetched from the `/Gui/Brightness` setting. Investigate whether gui-v2 is fetching the wrong min value here.

jpetrell, 2024-03-05:
Can no longer reproduce on v3.30~11 and the latest gui v2. Closing.

closed 2024-03-05 by jpetrell