victronenergy/gui-v2 #810: Generator overview box should be visible when no other AC inputs are connected

blammit, 2024-02-28:
Run venus-docker simulation "o". 

Observe on the Overview page that the left-handed box is often missing. Sometimes it will appear as "Shore" when the shore AC input is connected, other times as "Generator" when the generator is connected, then it will disappear occasionally. However, as per the design, if the grid/shore is not connected but a generator is present, the generator box should still be shown (even if it is not running/connected). In those cases, it can just show "Stopped" as the generator status.

blammit, 2024-02-28:
The generator box disappears because Connected=0 so gui-v2 assumes it should not be displayed in this case. But I think in this case it should be displayed anyway, to avoid having the empty space on the left hand side.

mpvader, 2024-02-28:
discussed in design meeting a while ago, making sure we finalise it here: https://victrondevelopment.slack.com/archives/C020QG3JVEW/p1709109365057129

blammit, 2024-03-01:
Design as been finalized. See "Solution A - Scale down font size" in slack discussion.

jpetrell, 2024-03-01:
Discussed with Serj that this may require touching the box size or paddings. When making the necessary modifications would maybe be good to fix some of the existing padding issues #730 at the same go.

closed 2024-03-27 by blammit