victronenergy/gui-v2 #853: Pages flicker during the opening and closing transitions

jpetrell, 2024-03-12:
Actual result: You can see the pages update and animations start and stop during the navigation, mostly notable case being the Overview page with bubble and connector animations. Depends on whether you open/leave the page with a swipe or from navigation bar, but sometimes the animation is started/stopped mid-transition (swipe in/out), sometimes it is stopped when the transition starts (leaving by tapping the nav bar), sometimes when the transition ends (opening by tapping the nav bar).

Expected result: To only display meaningful movement and calm down the transitions ideally the pages were layouted and the animations were started before the page slides into the view and paused only after leaving the page fully. Currently the animations are tied to the current page state true/false, but there could be additional activating and deactivating states also.

mpvader, 2025-06-05:
discussed in the meeting: should be super simple, so lets just do this

chriadam, 2025-06-19:
https://github.com/victronenergy/gui-v2/pull/2262 does this.

Other improvements will be investigated as part of issue #1938