victronenergy/gui-v2 #869: Page - VRM device instances

mr-manuel, 2024-03-16:
The page shows only the battery device instance, but all other are missing.

I have these devices connected and they are shown correctly in the device list:
- Battery
- PV Inverter
- Temperature Sensor (3x)
- MultiPlus II

[image]

[image]

jpetrell, 2024-03-26:
Didn't find any VRM device instances page on gui v1 to compare on so not sure how this should be implemented exactly.
Currently the VRM device instances page on gui v2 displays devices it finds under com.victronenergy.settings/Settings/Devices/* path, and VRM instance numbers from the /ClassAndVrmInstance value.

mpvader, 2024-03-27:
Not all device instances are configurable @mr-manuel .

@jpetrell : this is new functionality in gui-v2.  Not all device drivers in venus os support configurable instances. We might expand that over time, but its not at all a priority.

the requirement was to have device instances configurable for devices connected over VE.Direct. And they are now configurable - mission accomplished.

i’ll leave this issue open and on my name: the only remaining task is to document for users and installers how it works.