victronenergy/gui-v2 #873: DVCC "Auto selected" value and modbustcp list names cannot be fetched on wasm/MQTT

blammit, 2024-03-19:
There are two relevant bugs here when running gui-v2 via wasm/MQTT:

1. Go to Settings > DVCC. Observe the "Auto selected" shows "None" even when `mqtt/system/ActiveBmsService` points to a valid service. This text is supposed to show the value of `<service_uid/ProductName>` or `<service_uid/CustomName>` where `<service_uid>` is the value of `mqtt/system/ActiveBmsService`.

The problem is that the value of `mqtt/system/ActiveBmsService` is like a D-Bus service uid (like `com.victronenergy.battery.tty0`) instead of a MQTT service uid (like `mqtt/battery/123`). So, the composed uid becomes something like `com.victronenergy.battery.tty0/ProductName`, which are valid on D-Bus, but not on MQTT.

2. Go to Settings > Services > Modbus TCP. Enable Modbus TCP if needed, and click 'Available services'. Observe the first line in each list item shows the full service name, e.g. "com.victronenergy.battery", instead of the device's product name or custom name.

This is a similar problem to point 1. This text is supposed to show the value of `<service_uid/ProductName>`, where `<service_uid>` is a value under `mqtt/modbustcp/0/Services/<index>`, but this list of services contain strings like `com.victronenergy.battery.tty0`, which are valid on D-Bus, but not on MQTT.

Similar issues were resolved in https://github.com/victronenergy/gui-v2/issues/619.

To resolve 1), perhaps systemcalc could add a value like `mqtt/system/ActiveBmsInstance` so that gui-v2 can use that to compose `mqtt/<service>/<ActiveBmsInstance>`, similarly to how `mqtt/system/VeBusInstance` is provided alongside `mqtt/system/VeBusService`.

To resolve 2), perhaps systemcalc could add a list under `mqtt/modbustcp/0/ServiceInstances` or such, to provide a list of instance numbers that corresponds to the entries in ``mqtt/modbustcp/0/Services`.

chriadam, 2024-03-19:
Tagging @ReinvdZee as it involves systemcalc

blammit, 2024-03-19:
Tagging @izak who helped with resolving #619.

blammit, 2024-05-27:
Closing. Issue 1) will be resolved by #1025, so I've moved issue 2) to https://github.com/victronenergy/gui-v2/issues/1146.

closed 2024-05-27 by blammit