victronenergy/gui-v2 #909: Check if all Marine MFDs can show gui-v2 and come up with plan if not

mpvader, 2024-03-28:


jpetrell, 2024-06-12:
Drop 1.0.0 target?

mpvader, 2024-06-12:
No. We can't. Thanks for checking.

@ReinvdZee please make sure you take care of this, either this or next week. If you need any help, let me know.

ReinvdZee, 2024-06-13:
Tested this with this [Cerbo](https://vrm.victronenergy.com/installation/115349/overview), updated to V3.40~26

I added this line: 
`rewrite ^/$ /gui-beta redirect;`

to the end of `/etc/nginx/sites-enabled/default_server` (within last bracket)
and restarted nginx with `svc -t /service/nginx`. Now, nginx redirects requests to the remote-console to `gui-beta`.

Test systems:

| System | Software version information | Screen size | Results |
| --- | --- | --- | --- |
| Simrad NSS12 evo3S | Version 22.2 Application 66.1.9 | 1280 X 800 | Shows "loading" |  
| Simrad NSS12 evo3S | Version 22.2 Application 66.1.9 (Updated device) | 1280 X 800 | Shows "loading" |  
| Simrad GO9 XSE | Version 22.3 Application 66.2.13 | 800 X 480 | Shows "Open in a larger screen to view remote console" |  
| Garmin GPSMAP 8412 | Software version 34.00 | 1920 X 1200 | Shows "loading"
| Raymarine Axiom 12 | Software version 4.5.84 | 1280X800 | Works, full-screen and split-screen with other app. |

The smallest Simrad device won't boot properly, but I also don't expect it to work.

On the Raymarine MFD:
[image]
[image]

ReinvdZee, 2024-07-01:
@mpvader @jhofstee 

How do we handle this?

mman, 2024-07-22:
Some experiments here: https://github.com/victronenergy/venus-html5-app/pull/390

mpvader, 2024-10-04:
This taken care of, 99% and handled in issues elsewhere - closing this one

closed 2024-10-04 by mpvader