victronenergy/gui-v2 #911: Hide the Levels page from the main view if there are no tanks and environmental sensors

jpetrell, 2024-03-28:
Some customer configurations like vehicles almost always have tanks and sensors, but there are many land configurations, which have no use for tanks and sensors. In such configurations the Levels page will be totally empty and useless. Hide the Levels page when there are no tanks or sensors installed.

chriadam, 2024-04-01:
Err, who decided this?  Is this from mpvader or from jperez or from serj, or?  I didn't see any discussion about this in slack, but maybe I just missed it?  Or perhaps it was discussed in the meeting on Thursday?

DanielMcInnes, 2024-04-01:
> Err, who decided this? Is this from mpvader or from jperez or from serj, or? I didn't see any discussion about this in slack, but maybe I just missed it? Or perhaps it was discussed in the meeting on Thursday?

This was discussed in the meeting on Thursday

closed 2024-04-26 by DanielMcInnes