victronenergy/gui-v2 #917: Touch sometimes gets stuck after closing modal dialogs (wasm)

jpetrell, 2024-04-02:
Open GUI v2 from a browser. Go to Control Cards page. Repeatedly open and close generator dialog (e.g. tap "Manual Stop" to open the dialog, then "Close" to close the dialog, and repeat) or open and close Inverter / Charger dialog (e.g. tap "Mode: On", then "Close", and repeat).

Expected result: The dialogs open and close as expected.

Actual result: After couple of tries the touch of the whole GUI v2 gets stuck even though the dialog is no longer visible on the screen.

Seen on GUI v2 v0.2.14. Reproduced on both Chrome and Firefox. Looks like a Qt Quick Controls issue.

[Video demonstrating the issue](https://github.com/victronenergy/gui-v2/assets/2203667/1fd7354a-6e7d-4908-a62b-45b698f9bc7d)

jpetrell, 2024-04-02:
Debugged this. When the touch gets stuck getting "Uncaught RuntimeError: memory access out of bounds" or "Uncaught RuntimeError: index out of bounds" errors to the console, slightly different phrasing depending on the browser, but the same issue. According https://forum.qt.io/topic/154676/runtimeerror-index-out-of-bounds this seems to be caused by the app running out of stack. This seems to make sense since the error can be reproduced reliably (e.g. after the 3rd dialog opening) between the page refreshes.

jpetrell, 2024-04-02:
Building the wasm version with CMake config `target_link_options(VenusQMLModule PUBLIC -s STACK_SIZE=10MB`) seems to increase the number of dialog openings from 3-5 to 10-30 (desktop browsers) before the runtime error, but unfortunately doesn't fix the issue altogether.

I have been able to reproduce the issue on generator start/stop dialogs, minimum SOC dialog, time picker dialog and inverter / charged mode dialogs.

chriadam, 2024-04-03:
I'm not very familiar with WebAssembly, but it's curious to me that stack is being consumed here, given that these objects should be being allocated on the heap.  Or is it the case that in WebAssembly, heap operations are emulated via emplaced construction in a preallocated stack space?  I should read https://www.oreilly.com/library/view/webassembly-the-definitive/9781492089834/ch04.html properly, I suppose.  In any case, are we actually leaking dialog instances?  Do manual calls to gc() reduce the frequency of the issue?

DanielMcInnes, 2024-04-03:
This looks like https://bugreports.qt.io/browse/QTBUG-121382

jpetrell, 2024-04-03:
> This looks like https://bugreports.qt.io/browse/QTBUG-121382 , but the crash doesn't reproduce with the same example code. 

Yeah removing "modal: true" and/or exit transition at least does not help.

> it's curious to me that stack is being consumed here, given that these objects should be being allocated on the heap

Yeah good point. I probably assumed too much from limited test set.

jpetrell, 2024-04-03:
Found the reason for the crash, the WASM edition of the Qt UniformAnimator used in the press effect doesn't like to be destroyed in the middle of the animation:

```
    at memcmp (venus-gui-v2.wasm:0x1dd93e8)
    at QQuickShaderEffectPrivate::findMappedShaderVariableId(:6931/QByteArray const&) const (http://localhost:6931/venus-gui-v2.wasm)
    at QQuickShaderEffectPrivate::updateUniformValue(:6931/QByteArray const&, QVariant const&, QSGShaderEffectNode*) (http://localhost:6931/venus-gui-v2.wasm)
    at QQuickShaderEffect::updateUniformValue(:6931/QByteArray const&, QVariant const&) (http://localhost:6931/venus-gui-v2.wasm)
    at QQuickUniformAnimatorJob::updateCurrentTime(:6931/int) (http://localhost:6931/venus-gui-v2.wasm)
    at QAbstractAnimationJob::setCurrentTime(:6931/int) (http://localhost:6931/venus-gui-v2.wasm)
    at QParallelAnimationGroupJob::updateCurrentTime(:6931/int) (http://localhost:6931/venus-gui-v2.wasm)
    at QAbstractAnimationJob::setCurrentTime(:6931/int) (http://localhost:6931/venus-gui-v2.wasm)
    at QQmlAnimationTimer::updateAnimationsTime(:6931/long long) (http://localhost:6931/venus-gui-v2.wasm)
    at QUnifiedTimer::updateAnimationTimers(:6931/) (http://localhost:6931/venus-gui-v2.wasm)
```

Not related to PressEffect shader as such, happens also without any fragment shader. I will file a bug to Qt and before the proper fix implement a workaround.

chriadam, 2024-04-05:
Nice find.

PR #926 (2024-04-09): Avoid UniformAnimator crash on dialog closing (wasm)
Tried to stop(), complete(), pause() the UniformAnimator before the item deletion, but could still reproduce the crash so instead just avoid using the UniformAnimator on WebAssembly builds. Could alternatively have also added a bit of delay to the dialog destruction, but then could not really be sure the crash wouldn't reproduce in some other use case involving button presses and item deletions.

Fixes #917.
commit 4fb8162b (2024-04-09): Avoid UniformAnimator crash on dialog closing (wasm)

Fixes #917.