victronenergy/gui-v2 #921: Battery firmware version formatting issue

mpvader, 2024-04-04:
[image]

jpetrell, 2024-04-04:
Displayed PageDeviceInfo.qml by both v1 and v2, both use prefix + "/FirmwareVersion" value for displaying the firmware version. Looks like firmwareVersion(version, format) in ListFirmwareVersionItem.qml (GUI v2) formats the version incorrectly. There is a regular expression that tries to check if the value is hexadecimal, if not the version is returned as is. In this case looks like 1.26 is incorrectly interpreted as hexadecimal, though if I try setting 1.26 or "1.26" with dbus-spy the value already gets shown as "v1.26". Would be useful to know the exact source value and type.

There are bunch of fixes made to the function on February, maybe something regressed:
https://github.com/victronenergy/gui-v2/commit/8ed7b644f4efb60dba790b730b52a9afc6ed559f
https://github.com/victronenergy/gui-v2/commit/7974c248e40bab26703e4374bc224d8ff77b922a

@blammit you would be best to handle this as you have been iterating the version formatting here earlier and knows differences in how the devices report the firmware version?

blammit, 2024-04-05:
@jepefe any idea what the /FirmwareVersion value might be for this battery device? Looks like it should not be converted to hex and perhaps not to a major.minor string either, so maybe this is a special case that should be handled, similar to how `vebus` firmware values are handled specially?

@jpetrell the problem here is that in gui-v1, the firmware version comes directly from dbus as a text string that is already translated into hexadecimal etc, but in gui-v2, that is not available via MQTT, so we need to do the conversion at the app level.

blammit, 2024-04-05:
@jepefe is it possible for services to provide a `/FirmwareVersionText` value or such that provides the equivalent of the current `Text` value of `/FirmwareVersion` so that the UI layer doesn't need to translate the version string and do special handling for different device cases?

mpvader, 2024-05-27:
@jepefe pls answer this.

@ReinvdZee

ReinvdZee, 2024-05-29:
`can-bus-bms` fetches the firmware as big endian. It also fills the "text" value of the dbus path with `"v%d.%d"`. This string can be fetched with the `GetText` method. As discussed [here](https://github.com/victronenergy/venus/wiki/dbus-api#object-paths), that is how it is intended to work. 

I checked this briefly on [this](https://vrm.victronenergy.com/installation/33865/dashboard) system: 
The BMS reports the firmware in CAN frame 35F: 
`can0  35F   [8]  4C 69 04 17 C8 00 13 0E` (version: 0x04 0x17 --> v4.23)

`dbus-spy` reports `FirmwareVersion: 1047`, (= 0x0417). Pressing 't' in dbus-spy calls the `GetText` method and gives `v4.23`.

So Gui-v2 should just use the `GetText` method. Is this possible? Or maybe already the case? @blammit, @jpetrell?
If its already using `GetText`, then the issue may be somewhere else. 

**note** Unfortunately, this issue does not reproduce on the system I checked with. Gui-v2 shows the firmware version correctly.

jepefe, 2024-05-29:
The GetText method is available on DBus, but it cannot be called in the WebAssembly version that uses MQTT.

We have three firmware version formats.

```
VE.Direct/VE.Can/Generic         ->  Shown in the Device PageDeviceInfo
VE.Bus                           ->  Shown in the Device PageDeviceInfo
Venus                            ->  Shown in the firmware settings page
```
What we do in VictronConnect is check if `Mgmt/Connection == "VE.Bus"` then use vebus firmware format, otherwise use the VE.Direct/Generic one.

Venus firmware version is only shown in the firmware settings page so it does not need any extra logic to determine which format to use.

Doing the same in the gui-v2 should be enough.

jepefe, 2024-06-05:
In addition to the listed firmware version formats in my previous comment, there is an exception for the non-Victron Can Bus BMS.

```
// Note that /Mgmt/ProcessName for the Can BMS has been included in Venus v3.40~21.

if (/Mgmt/ProcessName == can-bus-bms) {
  snprintf(buf, len, "v%d.%d", (int)(version >> 8), (int)(version & 0xFF));
}
```

Please, add this exception to the fw version formatting function which I guess is this one:
https://github.com/victronenergy/gui-v2/blob/e165eb322893ea629ab51ea2029dff6a6ac905d9/components/settings/ListFirmwareVersionItem.qml#L12
This should solve the problem with the firmware version formatting for the BYD batteries and others.

@blammit

mr-manuel, 2024-06-06:
Will it be still possible to provide a string as version info like `v1.0.20240502dev` or is this going to be changed?

blammit, 2024-06-12:
> Will it be still possible to provide a string as version info like v1.0.20240502dev or is this going to be changed?

If supported by gui-v1 then it should be here as well, else it is a bug.

closed 2024-06-12 by blammit