victronenergy/gui-v2 #924: Improve app splash screen colour theme and text in browser/wasm

blammit, 2024-04-05:
Suggestions:

- Apply the light/dark theme depending on the system theme, rather than the light/dark theme configured in the application GUI settings? See related PR here: https://github.com/victronenergy/gui-v2/pull/792/files
- Don’t show technical detail
    -  but what if the error codes need to be seen for bug reporting etc?

podarok, 2024-04-05:
> but what if the error codes need to be seen for bug reporting etc?

There is a standard procedure of sending report from VictronConnect, could it work?

jpetrell, 2024-04-18:
> Apply the light/dark theme depending on the system theme, rather than the light/dark theme

This could be problem since GUI v2 itself always starts with dark theme, in the light case you would see the browser page flicker from white to black to white. Maybe we should initially always start with the dark theme also in the HTML splash screen to keep the color scheme changes at the minimum. Or we follow the system theme, but also pass the system theme info to GUI v2 to align the colors.

Discussed with Serj. Both splash screen cases should try to follow the same design that has already been defined in Figma mock-ups, no talk about Qt or WebAssembly, just the Victron Energy logo. In addition, some kind of loading indication is needed. On desktop the HTML splash screen appears on the screen only for a fleeting moment, but on mobile it can linger half-dozen seconds. "Loading..." label can be shown initially if the animating progress bar shown on the design (and GUI v2) is difficult to do with HTML. In the error case the label change shows an error, like in the Figma design.

mpvader, 2024-04-18:
Before you start on this, @jpetrell , which is not right now since prio :) , make sure to inform Dirk-Jan and Alex.

jpetrell, 2024-04-24:
> See related PR here: https://github.com/victronenergy/gui-v2/pull/792/files

When I access GUI v2 (wasm) on https://vrm.victronenergy.com the splash is already dark? The HTML splash I am seeing in the VRM website is different to what is defined in https://github.com/victronenergy/gui-v2/blob/main/.github/patches/index.html.patch . In addition to the splash the VRM website also shows a warning screen instructing the user to "Rotate device" if gui-v2 is accessed in the portrait screen orientation, which has light background, wonder if it should also default to dark.

jpetrell, 2024-04-25:
> The HTML splash I am seeing in the VRM website is different to what is defined

Ok, so learned that the user can also access the wasm build from local LAN network as long as his knows the GX device's IP address. So we have two HTML splash screens, one in the VRM website while WASM file is downloaded and the local patched one.

closed 2024-05-22 by jpetrell